linux-nvme/nvme-cli #879: "nvme discover" command crashes (seg. fault)

martin-belanger, 2021-02-01:
I'm using the command `nvme discover` with the option `--device` to retrieve discovery log pages for NVMe over Fabrics (i.e. TCP connections).

First, I create a persistent connection to a discovery controller with the following command and confirm that a new device got created, e.g. `/dev/nvme1`.
```
sudo modprobe -v nvme-tcp
sudo nvme discover --persistent -t tcp -a [IP Address] -s 8009
```

Then I try to re-use the same connection (i.e. device) to retrieve the discovery log pages (as shown below) and I get a seg. fault.
```
sudo nvme discover --device /dev/nvme1 -t tcp -a [Ip Address]
Segmentation fault
```

This was tested on Ubuntu 20.04 running Linux 5.8.

Another thing I noticed about the `--device` option is that the command still requires one to enter the `-t` and the `-a` options. If one reuses an existing connection, one should not have to specify the type (`-t`) and certainly not the IP address (`-a`). Right?

ericcurtin, 2021-02-09:
You want `sudo nvme discover --device nvme1 -t tcp -a [Ip Address]` I think

ericcurtin, 2021-02-10:
This prevents this from segfaulting in this case:

https://github.com/linux-nvme/nvme-cli/pull/901

ericcurtin, 2021-02-10:
#906 so this is documented.

ericcurtin, 2021-02-16:
Hey @keithbusch I think once we get #906 reviewed/merged for extra documentation, this should be good enough to close. What do you think?

hreinecke, 2021-10-06:
pull request is merged, so closing.

closed 2021-10-06 by hreinecke