linux-nvme/nvme-cli #889: nvme smart-log get "nan" data after glibc update

nebuchad908, 2021-02-04:
Hello.

OS: CentOS 7
nvme version: 1.8.1, 1.13.60

After update glibc to version 2.17-322.el7_9.x86_64, nvme-cli output strange values:

`unsafe_shutdowns                    : nan
media_errors                        : nan
num_err_log_entries                 : nan`

After downgrade to version 2.17-317.el7, all works fine:

`unsafe_shutdowns                    : 0
media_errors                        : 0
num_err_log_entries                 : 0`

keithbusch, 2021-02-04:
nan is short for "not a number". It's trying to print a 128 bit integer, but since there's no such thing, we calculate it as a long double, which should be able to capture the possible range of values close enough for human readability, and use the %lf format specifier accodingly. I wonder what changed in the newer glibc's printf() that isn't recognizing this format anymore.

nebuchad908, 2021-02-04:
> nan is short for "not a number". It's trying to print a 128 bit integer, but since there's no such thing, we calculate it as a long double, which should be able to capture the possible range of [...]

glibc 2.17-322 have a  minimum  37 changes about printf :)

Sn4kY, 2021-02-05:
same problem.
note that smartmontools wasn't affected, it still print correct output :
`# smartctl -a /dev/nvme0
Unsafe Shutdowns:                   0
Media and Data Integrity Errors:    0
Error Information Log Entries:      0
`

keithbusch, 2021-02-05:
> note that smartmontools wasn't affected, it still print correct output :

They're not using long floating point numbers with printf, though.

ericcurtin, 2021-02-06:
Changing this to print double's fixes it, unless you want to wait for a glibc patch?

https://github.com/linux-nvme/nvme-cli/pull/897
https://bugzilla.redhat.com/show_bug.cgi?id=1925204

Is double enough precision?

keithbusch, 2021-02-06:
Probably

keithbusch, 2021-02-06:
Just read the bugzilla, it's preferable that glibc picks up the fix on their end.

ericcurtin, 2021-02-06:
I closed the PR

igaw, 2022-01-19:
Closing as suggested in the last comment.