linux-nvme/nvme-cli #907: More questions about memory capacities of controllers

nxdiamond, 2021-02-10:
These are questions about the NVMe architecture not the Linux nvme-cli, but please forgive me because I can only deal with it using nvme-cli.

Suppose a computer has four NVMe controllers.

/dev/nvme0 doesn't support namespace management and it has 1TB capacity so nvme id-ctrl /dev/nvme0 reports tnvmcap is 0x0 instead of 0x10000000000.  I have a feeling that it's OK if /dev/nvme0 is unaware of the existence of other NVMe controllers.

/dev/nvme1 doesn't support namespace management and it has 1TB capacity so nvme id-ctrl /dev/nvme1 reports tnvmcap is 0x0 instead of 0x10000000000.  I have a feeling that it's OK if /dev/nvme1 is unaware of the existence of other NVMe controllers.

/dev/nvme2 and /dev/nvme3 do support namespace management and they have access to 4TB of capacity so I think it would be sensible if nvme id-ctrl /dev/nvme2 and dev/nvme3 should both report tnvmcap is 0x40000000000.

However, the spec for tnvmcap says "This field indicates the total NVM capacity in the NVM subsystem."  Should nvme id-ctrl /dev/nvme2 and dev/nvme3 both report tnvmcap is 0x60000000000, i.e. 6TB?  (I hope the answer is no.)

Suppose I tell /dev/nvme2 to create a namespace of capacity 4TB, so nvme id-ctrl /dev/nvme2 reports unvmcap as 0x0 because there is no unallocated space.  Then regardless of whether or not I tell /dev/nvm3 to attach the namespace, does nvme id-ctrl /dev/nvm3 has to report unvmcap as 0x0 because there is no unallocated space?  (I hope the answer is yes.)

Is it actually mandatory that /dev/nvme2 and /dev/nvme3 are aware of each other, and they both access the same pool of memory cells?  Is it actually impossible for /dev/nvme3 to have 1TB of private memory cells so it's impossible for /dev/nvme3 to report tnvmcap is 0x50000000000 and unvmcap is 0x10000000000?  (I hope the answer is yes, they access the same pool and it's impossible to have other private memory cells.)

I think there have to be three NVM subsystems in this computer.

Is it possible to add more NVMe controllers that do support namespace management, and is it possible for /dev/nvme4 and /dev/nvme5 to be aware of each other but to be unaware of the earlier ones, making a total of four NVMe subsystems?  (I hope the answer is no.)

keithbusch, 2021-02-10:
The only time controllers are aware of another is if they're in the same subsystem. The subsystem can contain any arbitrary number of controllers.

The TNVMCAP and UNVMCAP fields are global to the subsystem. If two controllers are in the same subsystem, they should be reporting the same values for each of these.

nxdiamond, 2021-02-10:
So it is possible for /dev/nvme0 to be in its own subsystem, /dev/nvme1 to be in its own subsystem, /dev/nvme2 and /dev/nvme3 to be in one subsystem which share 100% of their subsystem's capacity, and /dev/nvme4 and /dev/nvme5 to be in another subsystem which share 100% of their subsystem's capacity.  A Linux program can retrieve a list of six controllers but can it retrieve a list of four subsystems?

keithbusch, 2021-02-10:
The setup you've describe is allowed, and linux will be able to discover it. The `nvme list -v` command will break it out that way, too.

nxdiamond, 2021-02-10:
Sorry, I guess the command nvme list-subsys was staring me in the face and I didn't understand it until now.  Thank you.

closed 2021-02-16 by nxdiamond