linux-nvme/nvme-cli #914: Do Linux drivers cooperate with rescan?

nxdiamond, 2021-02-15:
nvme ns-rescan /dev/nvme6 tells the 6th controller to rescan its namespaces.  I guess if some other controller in the same subsystem is told to delete and create namespaces then nvme6 might not know about them, so a rescan makes sense.

But what about the Linux drivers and their collections of psuedo-files under /dev, /sys/block, etc.  Does the nvme ns-rescan command inform them of the changes?  Or do I have to perform more commands to make the kernel rescan the current configuration?

keithbusch, 2021-02-15:
In theory, if some other controller in the subsystem modifies another controller's namespace inventory, that controller should emit a namespace notify AER, so the manual rescan should never be necessary.

If anything changes, the kernel will make appropriate updates to sysfs and dev as needed during a rescan.

nxdiamond, 2021-02-16:
Sorry for essentially repeating the question you answered, but I want to double-check.  When a program or person invokes a command like nvme create-ns -s 1000000 -c 1000000 /dev/nvme6, and if the command succeeds, then the Linux kernel or driver will automatically be informed and it will create a device file something like /dev/nvme18n3 and other pseudo-files and directories under /sys/block, etc.?

Also if I use nvme delete-ns, the Linux kernel or driver will automatically be informed to delete all of the files related to the deleted namespace on all NVMe controllers that formerly had the namespace attached?

Is there any need to do things like echo 1 > /sys/bus/pci/devices/...something.../remove before nvme delete-ns?

(In comparison, parted has to invoke an ioctl to rescan partitions that can be created or deleted inside a namespace.)

keithbusch, 2021-02-16:
The create-ns command doesn't attach the namespace to any controller, so a rescan wouldn't find anything.

If you use attach-ns after creation, then that will show up after a rescan and the kernel will provide all the device handles and sysfs attributes for it. This should all happen automatically unless your kernel is really old. The same is true for delete-ns and detatch-ns commands.

You should never 'remove' a pci device that you want to manage with this tool. Removing the pci device will make the controller inaccessible.

nxdiamond, 2021-02-16:
The create-ns command has to specify a controller that gets told to create the namespace, so I didn't guess that the specified controller wouldn't get the namespace attached.  I understood that ns-attach would be needed for other controllers to use the same namespace, but didn't know that ns-attach would also be needed for the creating controller.

So, nvme ns-attach informs the kernel along with the hardware.  That is good, thank you.

On the other hand, if I intend to ns-delete a namespace, do I have to precede it with ns-detach commands to all of the controllers that currently have the namespace attached, or will the controllers and kernel figure it out from ns-delete alone?

keithbusch, 2021-02-16:
Spec says you should detach before deleting, but I don't think you are required to do that.

nxdiamond, 2021-02-16:
If the spec says it, I guess it's safer to do the detaches before deleting.  Thank you.

closed 2021-02-16 by nxdiamond