linux-nvme/nvme-cli #923: Does Linux understand a namespace attached to multiple controllers?

nxdiamond, 2021-02-19:
I can create a namespace on controller /dev/nvme6 and attach it, getting something like /dev/nvme4n18.  I can also attach it to controller /dev/nvme4 and get something like /dev/nvme3n2.  However, do Linux drivers and tools understand that /dev/nvme4n18 and /dev/nvme3n2 are the same block device?

If I use parted to create partitions /dev/nvme4n18p1 and /dev/nvme4n18p2, does Linux immediately create pseudo-files for /dev/nvme3n2p1 and /dev/nvme3n2p2?  If Linux doesn't immediately do so, then I have to call some ioctl to rescan the partitions in all nvme namespaces?

Also on the SATA planet, the mount command is smart enough that if I mount /dev/sda3 /mnt/sda3 and then try to mount /dev/sda3 /mnt/myusb then the second mount gives an error message.  But what happens when there are two pseudo-files with different names?  On the FibreChannel planet, when not using a controller that mediates redundant paths to drives in a drive array, Linux did not know that redundant paths are the same device so I could mount the same partition in two mount points and experience the obvious problems (therefore I had to use just one optical cable to the drive array).  Now in nvme, if I mount /dev/nvme4n18p1 /mnt/nvme4n18p1 and then try to mount /dev/nvme3n2p1 /mnt/nvme3n2p1, does the second mount know that I'm trying to mount the same partition at two mount points because the same namespace has both names and therefore give an error message?

(Sorry, I can't experiment by getting a computer with multiple nvme slots and multiple controllers that support namespaces.)

keithbusch, 2021-02-19:
The linux driver supports native multipathing. Unless you've disabled this, you will only see one namespace block device in /dev/, and the driver seamlessly handles controller path selection.

nxdiamond, 2021-02-19:
Good news, thank you.

On the other hand, this makes me think that block device names shouldn't be things like /dev/nvme19n222, but should just be things like /dev/nvmen222 and always let the driver handle controller path selection.

keithbusch, 2021-02-19:
The first number refers to the driver assigned subsystem instance, and we definitely want to see it in the name.

nxdiamond, 2021-02-19:
I see.  I thought I understood that /dev/nvme7n12 might be on a different controller than /dev/nvme7, but it's more complicated than that.  It's not just because two drivers can enumerate controllers in a different order.  It's because the 7 of /dev/nvme7n12 is a subsystem number not a controller number.

So that's also the reason why multipathing doesn't happen in the block device names.

keithbusch, 2021-02-19:
In your example, /dev/nvme7n12 might not be on controller /dev/nvme7 only if you have a really old kernel. The older naming was so confusing to pretty much everybody, so we changed it so that the namespace's first number (subsystem instance) could never match a controller instance from a different subsystem.

nxdiamond, 2021-02-20:
Yes it's confusing.  But I thought it still remains that way.  For example your comment in this issue:
https://github.com/linux-nvme/nvme-cli/issues/816

"Do you only have one nvme device? If not, make sure nvme0 really is the parent controller to nvme0n1. Sometimes it's not."

I guess you were concerned that the other person might have an old driver.

So after the change to coding and/or design of the drivers, suppose controllers 1 and 2 are in the first subsystem found and controller 3 is in the second subsystem found, then the first subsystem found could be assigned subsystem number 1 or 2 or 4 or 5 or 6 but not 3.  It's still confusing but it avoids some problems newwbies could cause by being too intuitive about the names of the pseudo-files they see in /dev.  Thank you for that.

closed 2021-02-24 by nxdiamond