linux-nvme/nvme-cli #929: fw-download command by default downloading complete firmware file

ganaidu707, 2021-02-22:
```
if (read(fw_fd, fw_buf, fw_size) != ((ssize_t)(fw_size))) {
        err = -errno;
        fprintf(stderr, "read :%s :%s\n", cfg.fw, strerror(errno));
        goto free;
}

while (fw_size > 0) {
        cfg.xfer = min(cfg.xfer, fw_size);
        
        err = nvme_fw_download(fd, cfg.offset, cfg.xfer, fw_buf);
        if (err < 0) {
	        perror("fw-download");
	        break;
        } else if (err != 0) {
	        nvme_show_status(err);
	        break;
        }
        fw_buf     += cfg.xfer;
        fw_size    -= cfg.xfer;
        cfg.offset += cfg.xfer;
}
```
in fw-download command by default reading the complete firmware and downloading in a loop, but as per the spec we have to download portion or complete fw file based on the NUMD(CDW10 ) value provided i.e **xfer** in tool?

Also 
```
fw_size = sb.st_size;
if ((fw_size & 0x3) || (fw_size == 0)) {
        fprintf(stderr, "Invalid size:%d for f/w image\n", fw_size);
        err = -EINVAL;
        goto close_fw_fd;
}
```

checking the alignment on fw-file size is not correct it seems?
we have to check align only on given NUMD is aligned or not?

correct me if I am wrong!

keithbusch, 2021-02-22:
> ```
> if (read(fw_fd, fw_buf, fw_size) != ((ssize_t)(fw_size))) {
>         err = -errno;
>         fprintf(stderr, "read :%s :%s\n", cfg.fw, strerror(errno));
>         goto free;
> }
>
> while (fw_size > 0) [...]

The xfer is in bytes, but the function converts it to dwords. Otherwise, I'm not sure what your question is.

> Also
>
> ```
> fw_size = sb.st_size;
> if ((fw_size & 0x3) || (fw_size == 0)) {
>         fprintf(stderr, "Invalid size:%d for f/w image\n", fw_size);
>         err = -EINVAL;
>         goto close_fw_fd;
> }
> ``` [...]

The command's size field is in DWORD units, so if the number of bytes in the file is not DWORD aligned, then there's no way to describe the correct transfer size in the command.

ganaidu707, 2021-02-23:
My question is we are suppose to download the firmware from given **offset** to **xfer** bytes, but we are reading whole firmware and downloading it in terms of **xfer.** So we are **not allowing a portion of the firmware to download, always complete FW file reading and downloading**. We shouldn't iterate in a loop over the **fw_size**, we have to iterate over  the **xfer**, until and unless user specified (may be we need to a extra flag for that). ?

another question is, the alignment we have to check it on **xfer** not on **fw_size**?

keithbusch, 2021-02-23:
The intention for the offset parameter was to account for firmware binaries spanning multiple files such that we have to inform the controller of the start for the next transfer, not to offset within the file.

The intention of xfer was how much to partial transfer with each command, but it's not the total. The reason for this is because controllers have max transfer limits, but fw download doesn't subscribe to MDTS, so we are defaulting to 4k per xfer because it's always safe.

We already enforce 4k aligned xfer, so no need to check it for dword alignment.

closed 2021-04-16 by ganaidu707