linux-nvme/nvme-cli #934: Is there a way to write file through nvme write

123toorc, 2021-02-24:
HI, according to issue #645, is there any way I can actually write to the file system through the nvme-cli write interface?

keithbusch, 2021-02-24:
You can write to any valid block on the namespace with this utility, but it's a bad idea to use this if you've formatted a filesystem on the namespace.

123toorc, 2021-02-24:
Thank you for such a quick reply!! 

Our project wants to add some features in the nvme protocol, specifically the admin command set. but to send a file directly to the block layer is not what we want. Is there anyway to let the sended file in the filesystem can be obverse?  

We are using nvme-of, we have mounted the remote nvme device on the client machine, I believe creating or copying a file to the mounted device, will this file io will trigger some nvme functions that includes the raw write or read of nvme-cli? 

Please help me! thank you so much!!!!!

keithbusch, 2021-02-24:
If you want to use a filesystem, don't use this tool for IO. Just use it like any other storage device.

123toorc, 2021-02-24:
Hi @keithbusch , Do you know how normal IO triggers the nvme IO? Do you know any related interfaces that NVME provides? Thank you!

keithbusch, 2021-02-24:
I'm not entirely sure I understand what your asking. The lower level protocols are usually not considered at the level of filesystem IO.

The Linux kernel provides several syscalls to issue read/write commands that are abstracted from the underlying storage protocols. These syscalls enter the through the kernel's virtual filesystem, and then forwarded to the actual filesystem implementation. The filesystem determines the location on disk to read/write the data and creates a generic "block IO" structure and sends that to the generic block layer. The block layer dispatches this to the low level driver, and that's where it enters the nvme driver. The nvme driver constructs an nvme read or write command from the generic block request, DMA maps the source/destination buffers, and dispatches that to the hardware.

The above is a very simplified version of what happens, and I'm leaving out a lot of the messy details.

closed 2021-03-09 by 123toorc