linux-nvme/nvme-cli #936: Write to NVM boot partition

ankitsingh1508, 2021-02-25:
Hi All,

Not sure if am making sense but i want to write something to nvm's boot partition.
Is it possible ? If yes then how ?

Regards,
AS

keithbusch, 2021-02-25:
```
# nvme fw-download <device> --fw <firmware-file>
# nvme fw-commit <device> --bpid=1
```

ankitsingh1508, 2021-02-26:
> ```
> # nvme fw-download <device> --fw <firmware-file>
> # nvme fw-commit <device> --bpid=1
> ```

ankitsingh1508, 2021-02-26:
Hi @keithbusch,

Thanks a lot for command help.
Was able to successfully copy the files to boot partition (fw-commit is not required as i believe fw-commit is to start the firmware update whereas my intention was just to copy files to boot partition). 

Is there a way to check the copied files ?

Regards,
AS

closed 2021-02-26 by ankitsingh1508

keithbusch, 2021-02-26:
The fw-commit should be required because that's the only command that specifies the boot partition as the destination for the downloaded imagine. Otherwise it's just in a temporary staging buffer.

This utility doesn't provide a way to read it back. I think it could be done for fabrics reliably, but pci won't work in most distributions since they prevent userspace mmio access when the driver is bound.

ankitsingh1508, 2021-02-26:
thanks @keithbusch ,
Actually my intention is to copy kernel self compiled bootable files and then have a code in pre-boot efi (will rename it to bootx64.efi for it to directly picked as bootable) to pick those kernel bootable files and boot the system.

Regards,
AS

keithbusch, 2021-02-26:
bpid=1 tells the controller that the staging buffer should be stored to the boot partition.

You can only write one firmware file at a time. You have to commit the most recently downloaded file before downloading another one, otherwise the second one will overwrite the first.

Only committed files are retained after a reboot.

ankitsingh1508, 2021-02-26:
So am assuming if fw-download and fw-commit command is used simultaneously then we can write multiple files (half of my problem solved..... :-) )
Since file is retailed after reboot, is there any way to delete them ?

keithbusch, 2021-02-26:
Not simultaneously. You have to use them sequentially.

You delete them by committing something else in its place.

Now I need to backup a second, because I gave some inaccurate statements earlier. The "bpid" parameter can be either 0 or 1, because controllers are supposed to support two boot partitions and that parameter selects which one. The correct way to say the downloaded image is destined for a boot partition is to use commit action 6.

ankitsingh1508, 2021-02-26:
Sorry for bothering you again and again, but thanks a lot for guidance.

I used bpid=0 as for my memory it showed '0' boot partition as supported/enabled.

But am bit confused, suppose I write 1st file (1.bin) by using download and commit sequentially and then write 2nd file (2.bin) again by using download and commit sequentially and so on. This way i can write N number of files still space is allowing.

So you mean to say 1.bin can be only removed by writing new 1.bin ?
How will I commit something else in the same place, as sequentially downloading and committing will keep on adding files ?

Regards,
AS

keithbusch, 2021-02-26:
There are two boot partitions: 0 and 1. You run fw-download to copy the file to the target, then fw-commit --action=6 to commit that image to a boot partition. You use the --bpid=0 to commit it to partition 0, or --bpid to commit it to partition 1. If you download another file and run commit, it will replace whatever was previously there in the specified partition.

ankitsingh1508, 2021-03-04:
Hi @keithbusch ,

Is it possible to read the file written to boot partition through 'nvme read' command? 

Regards,
AS

keithbusch, 2021-03-04:
No, the read command reads from logical blocks. The boot partition is stored somewhere else. Please read NVMe Specification section 8.13.1.

ankitsingh1508, 2021-03-08:
Hi @keithbusch,

Thanks for referring me to "8.13.1 Reading from a Boot Partition" of NVMe Specification.
It saying reading DATA from boot partition is possible.

Will try it out.

Regards,
AS