linux-nvme/nvme-cli #955: nvme delete-ns was blocked on Intel P4610

stutiredboy, 2021-03-03:
after detach-ns, while try to delete-ns, the command was blocked.

```
# nvme delete-ns /dev/nvme1 -n 1
```

1. I try to add `--timeout=xxx`, but it has no help;
2. I can't kill nvme command, even I use `kill -9`;
3. after I reboot the OS, the namespaces was deleted;
4. I tried nvme version 1.7 and 1.13;
5. strace log as below:

```
# strace ./nvme delete-ns /dev/nvme1 -n 1 --timeout=1000
execve("./nvme", ["./nvme", "delete-ns", "/dev/nvme1", "-n", "1", "--timeout=1000"], 0x7fff23680db8 /* 21 vars */) = 0
brk(NULL)                               = 0x5651e7666000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=28340, ...}) = 0
mmap(NULL, 28340, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f66795f3000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\260A\2\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=1824496, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f66795f1000
mmap(NULL, 1837056, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f6679430000
mprotect(0x7f6679452000, 1658880, PROT_NONE) = 0
mmap(0x7f6679452000, 1343488, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x22000) = 0x7f6679452000
mmap(0x7f667959a000, 311296, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x16a000) = 0x7f667959a000
mmap(0x7f66795e7000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1b6000) = 0x7f66795e7000
mmap(0x7f66795ed000, 14336, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f66795ed000
close(3)                                = 0
arch_prctl(ARCH_SET_FS, 0x7f66795f2540) = 0
mprotect(0x7f66795e7000, 16384, PROT_READ) = 0
mprotect(0x5651e5f80000, 8192, PROT_READ) = 0
mprotect(0x7f6679621000, 4096, PROT_READ) = 0
munmap(0x7f66795f3000, 28340)           = 0
brk(NULL)                               = 0x5651e7666000
brk(0x5651e7687000)                     = 0x5651e7687000
openat(AT_FDCWD, "/usr/lib/locale/locale-archive", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=3031632, ...}) = 0
mmap(NULL, 3031632, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f667914b000
close(3)                                = 0
openat(AT_FDCWD, "/dev/nvme1", O_RDONLY) = 3
fstat(3, {st_mode=S_IFCHR|0600, st_rdev=makedev(0xf6, 0x1), ...}) = 0
ioctl(3, NVME_IOCTL_ADMIN_CMD
```

keithbusch, 2021-03-03:
Looks like a kernel issue. What kernel version are you using?

stutiredboy, 2021-03-03:
kernel information: `Linux hostname 4.19.0-14-amd64 #1 SMP Debian 4.19.171-2 (2021-01-30) x86_64 GNU/Linux`

4.19.0

keithbusch, 2021-03-03:
Ok, that's very old. Would you be able to see if a newer version works better? Maybe a 5.10? I'll check tomorrow if we need a stable patch too 4.19.

stutiredboy, 2021-03-03:
thanks for your help.

4.19 is the newest kernel on Debian Release.

It works after I update the kernel to 5.10.

keithbusch, 2021-03-03:
4.19 stable needs https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit?id=6abff1b9f7b8884a46b7bd80b49e7af0b5625aeb.

hreinecke, 2021-10-09:
Guess we can close it as it relates to an older version (and a link to the patch has been provided).

closed 2021-10-09 by hreinecke