linux-nvme/nvme-cli #959: nvme id-ns json output lacks "in use" marker

tonecodez, 2021-03-04:
```
$ ./nvme id-ns /dev/nvme1n1
...
lbaf  0 : ms:0   lbads:9  rp:0 (in use)
```

vs 

```
$ ./nvme id-ns /dev/nvme1n1 -o json
...
  "lbafs" : [
    {
      "ms" : 0,
      "ds" : 9,
      "rp" : 0
    }
  ]
```

Want to make sure of a couple things:
1. Is this expected?
2. Will -o json only show lbafs that are 'in use'?

tonecodez, 2021-03-05:
maybe another field 
"in-use" : 1 for -o json would be helpful.

Again, not entirely sure how this mech works; any insight would be helpful.

keithbusch, 2021-03-05:
Sure, I suppose it could be added. I think the intention for json was that it'd be parsed by another program, and it'd be pretty trivial for such a program to find the in-use format. It should be safe to add the field into the output for more human friendly, though.

tonecodez, 2021-03-05:
> it'd be pretty trivial for such a program to find the in-use format.
can you give an example of this? can this be done with the json output alone?

keithbusch, 2021-03-05:
The "flbas" field provides the index of the "in-use" lbaf.

tonecodez, 2021-03-05:
Thanks, will close issue.

closed 2021-03-05 by tonecodez