linux-nvme/nvme-cli #975: [Question] Setting Arbitration Mechanism

YuriGrigorov, 2021-03-19:
Hi!

I have some NVME devices (Huawei HWE52P438T0L005N, Intel SSDPE2KX080T8)

For both of them `nvme show-regs` says Weighted Round Robin Arbitration is supported

```
# nvme show-regs -H /dev/nvme0 | grep "(AMS)"
	Arbitration Mechanism Supported  (AMS): Weighted Round Robin with Urgent Priority Class is supported
	Arbitration Mechanism Selected     (AMS): Round Robin
```

I'd like to do some tests with I/O Prioritization, but absolutely have no idea how to change AMS from RR to WRR.

```
release: CentOS Linux release 7.6.1810 (Core)
kernel: 5.4.84-3.el7.x86_64
nvme-cli version: 1.6
```

keithbusch, 2021-03-19:
The driver has to set this up, and this particular command just shows what the driver did. The linux driver never enables it, though. The WRR feature was very poorly defined and is a poor match for the kernel io scheduler, so the driver doesn't even try to enable it.

closed 2021-03-22 by YuriGrigorov

yangliu7999, 2024-04-17:
> The driver has to set this up, and this particular command just shows what the driver did. The linux driver never enables it, though. The WRR feature was very poorly defined and is a poor match for the kernel io scheduler, so the driver doesn't even try to enable it.

Hello, I am currently trying to use the WRR (Weighted Round Robin) feature. I have configured the nvme driver according to the code in D2FQ. However, during my usage, I have not observed the SSD allocating different bandwidths to queues based on their weights. I would like to know how to resolve this issue. Thank you! The SSD I am using is Intel P5530 960G, and the OS is Ubuntu 20.04.

yangliu7999, 2024-04-17:
> D2FQ is new to me. Google says it's something built on top of WRR and some papers have been released
>  by the Sungkyunkwan University. I was not able to find anything related in the kernel. This means the situation has not changed in the upstream code, so you are on your own here. I suggest you contact the authors of the D2FQ algorithm, maybe they are able to help you.
>
> Please also reviving old threads/issues. You can reference them in a new issue if needed. This avoids cluttering exiting reports. Thanks.

Thanks！