linux-nvme/nvme-cli #986: Read and Print PCIe Registers

ganaidu707, 2021-03-26:
Reading and printing` PCIe registers` in nvme-cli just like `controller registers`, will it be a good idea?

keithbusch, 2021-03-26:
Most linux distributions won't let user space access pci registers anymore, so we can't rely on it.

keithbusch, 2021-03-26:
I assumed you meant io registers, but do you mean config registers? If so, there's no need to duplicate that in this tool because lspci and setpci already have capability that covered.

ganaidu707, 2021-03-26:
using `lspci -s B:D,F -x` or `setpci -s B:D.F reg+off.size` read the dump and print?

keithbusch, 2021-03-26:
Yeah, those tools already print out config space registers in a nice format. Is there something different you want them to do?

ganaidu707, 2021-03-26:
okay, I think we both commented at the same point in time.
yes, those tools give the dump in a nice format, can we print it in a means that we do it for the other parameters in "`normal and JSON"` format?

keithbusch, 2021-03-26:
Not sure how you can json'ify output from setpci. There's nothing indicating what register you're printing:

```
# setpci -s 09:00.0 0.w
15b7
```

There is interest in json output for lspci, though, but lspci should handle it because that's operating at a protocol level lower than nvmecli wants to maintain.

ganaidu707, 2021-03-26:
```
static int nvme_pcie_header_cap(int argc, char **argv,    
       struct command *cmd, struct plugin *plugin)
{
    const char *desc = "Retrieve PCIe Header Registers from "\
		"offset 0x0 to 0x3f, input is BDF, bus, device and "\
		"function, can be retried using lspci command";
	const char *bus = "Bus Number";
	const char *device = "Device Number";
	const char *function = "Funcion Number";
    const char *fmt = "output format json|normal";
    int fd, err = 0;
	char bdf[8];
    FILE *fp;
	char *header = NULL;
	uint8_t output[64];
    enum nvme_print_flags flags;

    struct config {
		int bus;
		int device;
		int function;
		char *output_format;
	};
    
    struct config cfg = {
		.bus = 0,
		.device = 0,
		.function = 0,
        .output_format = "normal",
    };

    OPT_ARGS(opts) = {
		OPT_UINT("bus", 	     'b', &cfg.bus, 			bus),
		OPT_UINT("device", 		 'd', &cfg.device, 			device),
		OPT_UINT("function", 	 'f', &cfg.function, 		function),
        OPT_FMT("output-format", 'o', &cfg.output_format, 	fmt),
        OPT_END()
    };

    err = fd = parse_and_open(argc, argv, desc, opts);
    if (fd < 0)
        goto ret;

    err = flags = validate_output_format(cfg.output_format);
	if (flags < 0)
		goto close_fd;

	header = calloc(64, 1);
	
	printf("bus = %d, device = %d, function = %d\n", cfg.bus, cfg.device, cfg.function);
	sprintf(bdf, "%x:%x.%x", cfg.bus, cfg.device, cfg.function);
	printf("bdf = %s\n", bdf);
	sprintf(command, "a=0; while [ $a -lt 64 ]; do setpci -s %s $a.B; a=`expr $a + 1`; done", bdf);
	printf("command = %s\n", command);
	fp = popen(command, "r");
	if (fp == NULL) {
		printf("failed to retrieve header registers\n");
		goto close_fd;
	}
	int i = 0;
	while(fgets(header, sizeof(header), fp) != NULL) {
		output[i] = (__u8)strtol(header, NULL, 16);
		printf("> i = %d -> Reg %s <\n", i, header);
		i++;
	}
	
	for (i = 0; i < 64; i++)
		printf("> i = %d -> output %u <\n", i, output[i]);

	pclose(fp);

	nvme_show_pice_header_regs(output);

    free(header);

close_fd:
	close(fd);
ret:
	return nvme_status_to_errno(err, false);
}

```
I have prepared shell command to read up to 64 bytes (byte by byte)u and executed using `popen`
This how I am going to collect the dump and in `nvme_show_pice_header_regs` going to print it in `JSON` format based on offsets given nvme-spec.

This for the PCI Header config up to offset 0x3f, if you like I will send it PR after clean up.

keithbusch, 2021-03-26:
No, please don't send this. It has nothing to do with nvme and is equally applicable to any pci end device. Therefore it doesn't belong here.

ganaidu707, 2021-03-26:
Okay, thanks

closed 2021-03-26 by ganaidu707