linux-nvme/nvme-cli #987: No info for NVME WD black SN750

tifDev, 2021-03-30:
Hello,

I bought a NVME WD black SN750. 
I've added a msdos partition and then formated to ext4

I've installed then the nvme-cli

Tried to do some tests:
```
sudo nvme list
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev  
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------

sudo nvme smart-log /dev/sdc1
smart log: Invalid argument
```

Nothing gets displayed and nothing works.

May I know the reason pls?

Thank you.

birkelund, 2021-03-30:
How did you partition and format it?

keithbusch, 2021-03-30:
This is confusing. Why are you running nvme commands to /dev/sdc? That's not an nvme drive.

tifDev, 2021-03-31:
> How did you partition and format it?

With gparted I created a partition table msdos for the whole disk space and then formatted to ext4.

> Why are you running nvme commands to /dev/sdc? That's not an nvme drive.
@keithbusch how can you state this ? :)
My NVME got mounted on my system on /dev/sdc1

birkelund, 2021-03-31:
> > How did you partition and format it?
> 
> With gparted I created a partition table msdos for the whole disk space and then formatted to ext4.
> 
> > Why are you running nvme commands to /dev/sdc? That's not an nvme drive.
> > @keithbusch how can you state this ? :)
> > My NVME got mounted on my system on /dev/sdc1

Then it is not an NVMe drive ;)

NVMe drives shows up as `/dev/nvmeXnY`.

tifDev, 2021-03-31:
How should I format it then? Because this is a NVME (WD black SN750)? 

**Note:** I'm currently connecting it with an adapter for USB3.0

birkelund, 2021-03-31:
> **Note:** I'm currently connecting it with an adapter for USB3.0

**BINGO** - This means that the kernel sees a USB mass storage device, *not* and NVMe device ;)