linux-nvme/nvme-cli #989: More questions about namespaces, controllers, and subsystems

nxdiamond, 2021-03-30:
Thank you Mr. Busch for many kind answers.  Sorry to ask more questions now.  I don't have enterprise grade devices so can't experiment.

/sys/block/nvme1n2/device is a symbolic link to a controller's directory somewhere below /sys/, say for example nvme3.  If subsystem 1 contains several controllers and a few of them have namespace 2 attached, I guess the symbolic link will point to one of those controllers and I don't have to care which one.  But suppose namespace 2 is orphaned, not attached to any controller, then does the driver delete the symbolic link entirely?

nvme id-ctrl /dev/nvme1n2 works (I was expecting an error message), though I could only experiment on an emulated device in VMware.  I guess the command or driver chooses a controller in subsystem 1, maybe a controller that has namespace 2 attached?

nvme smart-log /dev/nvme1n2 works.  But I guess the statistics must be for all of the memory in subsystem 1, not limited to namespace 2 or one of the controllers?

nvme smart-log /dev/nvme3 works.  Again I guess the statistics must be for all of the memory in the subsystem that contains controller 3, not specific to controller 3?

nvme list -v outputs three sections of information.  In the third section, where each line displays a namespace, there is a list of controllers at the end.  Is this a list of controllers that are in the same subsystem as the namespace (i.e. it is probably possible to attach the namespace to each controller in the list) or is this a list of controllers that actually already have the namespace attached?

In the second section of output from nvme list -v, where each line displays a controller, I have a feeling that a bug causes some information not to be displayed that was intended to be displayed.

keithbusch, 2021-03-30:
The driver doesn't enumerate namespaces that are not attached to a controller, so no sysfs entries for those.

You can always send admin commands through namespace handles, so id-ctrl to /dev/nvme1n2 is fine. A controller path will be chosen by the driver to dispatch the command.

The scope of contents of smart log are dependent on controller capabilities, whether they're specific to the namespace or subsystem.

The list -v output was supposed to list only attached controllers. Some versions of this tool have a bug where it lists all known controllers in the subsystem whether attached to the namespace or not.

keithbusch, 2021-03-30:
If it's possible to attach a namespace to multiple controllers, then it's multipath capable. If you're using a multipath capable kernel (which you should if you have multipath capable namespaces), then the "device" link should be pointing to the subsystem rather than a specific controller.

closed 2021-04-26 by nxdiamond