linux-nvme/nvme-cli #999: Makefile: lib existence check does not work in centos7

sc108-lee, 2021-04-05:
Because of below problem, Makefile always try to build with libs, even though they are not exist on system.

[root@localhost ~]# ld --version
GNU ld version 2.27-41.base.el7
Copyright (C) 2016 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.
[root@localhost ~]# ld -o /dev/null -ljson-c
ld: warning: cannot find entry symbol _start; not setting start address
[root@localhost ~]# echo $?
0

[root@localhost ~]# scl enable devtoolset-9 bash
[root@localhost ~]# ld --version
GNU ld version 2.32-16.el7
Copyright (C) 2019 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.
[root@localhost ~]# ld -o /dev/null -ljson-c
ld: warning: cannot find entry symbol _start; not setting start address
[root@localhost ~]# echo $?
0

sc108-lee, 2021-04-05:
[root@localhost nvme-cli]# make
NVME_VERSION = 1.9.558.g7ade
    CC nvme-print.o
In file included from nvme-print.c:8:0:
nvme.h:26:25: fatal error: json-c/json.h: No such file or directory
 #include <json-c/json.h>
                         ^
compilation terminated.
make: *** [nvme-print.o] Error 1

Of course, I will be work after installing that lib on the system.

igaw, 2022-01-07:
With the meson build system we should covered this now.