linux-nvme/nvme-cli #1004: Guarantee the writing order in the Zoned Namespace of the NVMe device

ryderhsu0129, 2021-04-07:
I want to vrify Zoned Namespace feature of NVMe SSD through nvme-cli.
If I send two write commands to the Zoned Namespace at the different time(but time is close),
can it be guaranteed that the controller executes in order without errors?

I found that the I/O scheduler mq-deadline has an implementation(Zone Write Locking) to control write sequence,
but it seems that it's only used in SATA device, not in the NVMe drive.
So is there a chance that an error will occur on NVMe device because the writing operations is not in order?

thanks!

keithbusch, 2021-04-07:
We can't guarantee the execution order if you dispatch multiple commands concurrently.

ryderhsu0129, 2021-04-07:
> We can't guarantee the execution order if you dispatch multiple commands concurrently.

@keithbusch 

thanks for your reply.

Can I ask another question?
so there is no mechanism in the kernel to prevent out-of-sequence write operations on ZNS of NVMe device?
like SCSI uses the I/O scheduler mq-deadline that has an implementation Zone Write Locking to control write sequence.

keithbusch, 2021-04-07:
Passthrough commands (which are the only kind of commands that nvme-cli will dispatch) always bypass the scheduler, so the zone write locking doesn't apply to this particular tool.

closed 2021-04-08 by ryderhsu0129