linux-nvme/nvme-cli #1010: need to set timeout for nvme 1.4 get lba status command

robinejoh, 2021-04-15:
Hello,

According to NVMe 1.4 spec,
5.27 Get LBA Status command

b)	Untracked LBAs: a list of Potentially Unrecoverable LBAs generated by a scan originated by a Get LBA Status command with the ATYPE field set to 10h. The controller scans internal data structures related to LBAs to determine which LBAs are Potentially Unrecoverable LBAs. The controller may use this scan to determine which LBAs in which namespaces are affected by a component (e.g., die or channel) failure. **Significant delays may be incurred** during the processing of a Get LBA Status command with the ATYPE field set to 10h. After discovery of Untracked LBAs, they may or may not be added to the list of Tracked LBAs.

"Significant delays" means that it may take longer than usual nvme commands to complete get lba status (Atype 10h) command.
So, in my opinion host need a feature to set desired timeout when it issue get lba status command.

keithbusch, 2021-04-15:
Sounds reasonable. Parameter added (it will default to the system default).

robinejoh, 2021-04-16:
@keithbusch , 
Thanks, I will close the issue.

closed 2021-04-16 by robinejoh