linux-nvme/nvme-cli #1021: Intel Megaraid SAS controller + NVMe device

nxdiamond, 2021-04-29:
Sorry this is a driver problem but I don't see an "Issues" tab in page https://github.com/torvalds/linux/tree/master/drivers/nvme

PCI vendor 8086 device 2822 is an Intel SATA RAID controller and is partially handled by the Linux SATA AHCI driver.  However, I have the impression that Intel modified a Megaraid driver to add support for NVMe.  But I don't know if the modified Megaraid driver supports that particular chip ... compounded by the impression that there are more than one model of chip using that PCI identifier.

A friend has an ASUS PC with that SATA RAID controller attached to 4 devices:  SATA DVD drive, SATA HDD, SATA SSD, and NVMe SSD.  Windows 10 accesses all drives without any problem.  The NVMe drive has data on it.  However, Linux only sees first 3 devices.  Linux thinks the NVMe drive doesn't exist.  There is no sysfs entry.  Command "nvme list" doesn't show any subsystems or controllers or namespaces.

Do you know how to make Linux recognize NVMe devices attached to that kind of controller?

keithbusch, 2021-04-29:
You have to disable RAID mode in your system BIOS. This is a proprietary mode that Intel refuses to open, so no Linux support for the AHCI remmaped NVMe device in Linux.

nxdiamond, 2021-04-29:
Even though none of the drives are RAIDed (JBOD, just bunch of disks), the chip itself behaves differently depending on the UEFI settings?

If he can turn off the setting that makes the chip capable of RAID, then the chip will do transparent passthrough of the NVMe device, instead of current behaviour that remaps the device to a something that's accessible to the Windows AHCI driver but inaccessible to the Linux AHCI driver?

I can't actually ask him to turn off the setting in his UEFI because if the chip behaves differently then I bet the next time he boots his installed Windows system it won't even find the drive it's installed on.  More importantly, I can't tell customers to change their UEFI settings like that.  I'll have to just tell them that Intel hates Linux sometimes.

keithbusch, 2021-04-29:
"RAID" mode is just the name of the mode. It doesn't require you actually stripe the devices. The effect of enabling it just means that separate storage devices are aggregated behind a special AHCI controller that only Intel RST knows how to see what is on the other side. If you disable the mode, then all the devices are still visible to Windows, but would bind to whatever default driver is provided by the OS.

Why can't you recommend disabling the option? That's exactly what all the OEM's that ship with this option require. If you actually want to use the RST driver in Windows, then just turn the option back on.

nxdiamond, 2021-04-29:
'"RAID" mode is just the name of the mode. It doesn't require you actually stripe the devices.'

Yes I know.  In the BIOS or UEFI it's just settings and the driver gets to decide whether to obey the settings or do JBOD.  (Even in SCSI RAID (hardware RAID) you can do JBOD but then the driver doesn't make the decisions.)

'Why can't you recommend disabling the option?'

It's not the kind of thing we could impose on most of our customers.

Intel's special mode lets libata see actual SATA DVD, HDD, and SSD drives on the other side, so I wonder why they didn't make their NVMe-to-SATA mapping transparent.  Anyway thank you for explaining what they did though I don't expect you to know why.

nxdiamond, 2021-04-29:
There's no solution for me but there is for some people ('Why can't you recommend disabling the option? That's exactly what all the OEM's that ship with this option require for Linux.')  So, neither solved nor unsolvable but we're stuck with it.  Thank you.

closed 2021-04-29 by nxdiamond