linux-nvme/nvme-cli #1024: NVMe status: INVALID_OPCODE: The associated command opcode field is not valid(0x4001)

MilesBHuff, 2021-05-05:
When issuing `nvme delete-ns -n 1 /dev/nvme0`, I get: `NVMe status: INVALID_OPCODE: The associated command opcode field is not valid(0x1)`.  This is for a WD Black.  I can find no way to update its firmware.

When issuing `nvme delete-ns -n 1 /dev/nvme1`, I get: `NVMe status: INVALID_OPCODE: The associated command opcode field is not valid(0x4001)`.  This is for a Samsung EVO.  Its firmware is up-to-date, but was released 2 years ago.

From what I've seen in other issues here, it sounds like the above are probably firmware issues.  However, it seems ¿wrong? that *both* my NVMe drives would fail the same operation like this.  I decided to post this issue in case these somehow *aren't* firmware issues, or in case you know something I can try.

I'm trying to delete the default namespace on each drive, and create 2 namespaces on each:  one to give to ZFS, and one to use for the ESP and swap.  (ZFS behaves differently (better) when given a whole disk.)

birkelund, 2021-05-05:
I'm pretty sure none of those devices support Namespace Management. Try

    # nvme id-ctrl /dev/nvme0 -H | grep 'NS Management'

MilesBHuff, 2021-05-05:
Ah, bless.  Thanks for the extremely quick reply.  Sure enough, when I run that on each drive, I get : `  [3:3] : 0     NS Management and Attachment Not Supported`.  Guess I'll just make due without.

closed 2021-05-05 by MilesBHuff