linux-nvme/nvme-cli #1037: smart-log reports errors but none found in error-log

jouir, 2021-05-15:
I have bought a brand new Western Digital 2TB NVMe but nvme-cli reports errors with the `smart-log` action:
```
Smart Log for NVME device:nvme0 namespace-id:ffffffff
critical_warning                        : 0
temperature                             : 42 C (315 Kelvin)
available_spare                         : 100%
available_spare_threshold               : 10%
percentage_used                         : 3%
endurance group critical warning summary: 0
data_units_read                         : 201,526,305
data_units_written                      : 188,048,213
host_read_commands                      : 660,948,177
host_write_commands                     : 251,748,228
controller_busy_time                    : 4,612
power_cycles                            : 6
power_on_hours                          : 241
unsafe_shutdowns                        : 0
media_errors                            : 0
num_err_log_entries                     : 1
Warning Temperature Time                : 0
Critical Composite Temperature Time     : 0
Thermal Management T1 Trans Count       : 0
Thermal Management T2 Trans Count       : 0
Thermal Management T1 Total Time        : 0
Thermal Management T2 Total Time        : 0
```
1 error reported:
```
# ./nvme smart-log /dev/nvme0 | grep num_err_log_entries
num_err_log_entries                     : 1
```

But `error-log` doesn't report any error:
```
# ./nvme error-log /dev/nvme0 -o json | jq "[.errors[].error_count] | add"
0
```

Full output:
```json
{
  "errors" : [
    {
      "error_count" : 0,
      "sqid" : 0,
      "cmdid" : 0,
      "status_field" : 0,
      "phase_tag" : 0,
      "parm_error_location" : 0,
      "lba" : 0,
      "nsid" : 0,
      "vs" : 0,
      "trtype" : 0,
      "cs" : 0,
      "trtype_spec_info" : 0
    },
    {
      "error_count" : 0,
      "sqid" : 0,
      "cmdid" : 0,
      "status_field" : 0,
      "phase_tag" : 0,
      "parm_error_location" : 0,
      "lba" : 0,
      "nsid" : 0,
      "vs" : 0,
      "trtype" : 0,
      "cs" : 0,
      "trtype_spec_info" : 0
    },
[... 855 lines cut ...]
      "error_count" : 0,
      "sqid" : 0,
      "cmdid" : 0,
      "status_field" : 0,
      "phase_tag" : 0,
      "parm_error_location" : 0,
      "lba" : 0,
      "nsid" : 0,
      "vs" : 0,
      "trtype" : 0,
      "cs" : 0,
      "trtype_spec_info" : 0
    }
  ]
}
```

IMHO, there is an inconsistency between `smart-log` and `error-log`.

Version:

```
# ./nvme --version
nvme version 1.14.13.ge400
```

keithbusch, 2021-05-15:
The tool just reports what the drive tells us. If you think the output is incorrect, then you should report the issue to the device vendor.

igaw, 2022-01-19:
Question got answered. Closing issue.

closed 2022-01-19 by igaw