linux-nvme/nvme-cli #1048: DSM with multiple namespaces

ahribeng, 2021-05-19:
When an nvme device has multiple namespaces how should I construct the command for DSM?

Option 1 ( This works for me)
nvme dsm /dev/nvme0nX -a 0 -s 0 -b 100663296 -d

Option 2 ( This does not work for me, but would prefer it if valid)
nvme dsm /dev/nvme0 -a 0 -s 0 -b 100663296 -d -n X

keithbusch, 2021-05-19:
The spec defines the dsm command as targeting a single namespace. You'll need to construct a dsm for each namespace individually.

The Linux kernel doesn't let you send IO commands through the admin handle anymore for multi-namespace controllers, so /dev/nvme0 won't  work.

ahribeng, 2021-05-19:
okay thanks

ahribeng, 2021-05-19:
Does it make sense to remove the -n option from the command or is there still a use case for it?

keithbusch, 2021-05-19:
I guess the option still works on older kernels, or ones that have just a single attached namespace. Or you could use it to override the namespace id of the block handle on most kernels too, though 5.13 won't let you do that.

ahribeng, 2021-05-19:
okay was going to offer a pull request to remove or change documentation, but seems that the usage is still valid. I will go ahead and close this.

closed 2021-05-19 by ahribeng