linux-nvme/nvme-cli #1049: virt-mgmt Number of Resources always 0 using v1.14.13.ge400

theknowledge, 2021-05-21:
Virtualization Management commands are broken in nvme-cli v1.14.13.ge400

Issuing commands:
~# nvme virt-mgmt /dev/nvme0 -c 0x0001 -n 5 -r 0 -a 8
success, Number of Resources allocated:0x0
~# nvme virt-mgmt /dev/nvme0 -c 0x0001 -n 5 -r 1 -a 8
success, Number of Resources allocated:0x0
~# nvme virt-mgmt /dev/nvme0 -c 0x0001 -a 9
NVMe status: INVALID_SECONDARY_CTRL_STATE: The action requested for the secondary controller is invalid based on the current state of the secondary controller and its primary controller.(0x120)

No resources have been assigned.
I checked the commands using a PCIe analyser, and the reason is that the number of resources requested (NR field) appears as 0 on a PCIe bus.
[image]

Works fine using nvme-cli v1.9
[image]

keithbusch, 2021-05-21:
Thank you for the notice, I've pushed a fix for the reported bug.

And just for future reference since most people won't have a protocol analyzer handy, we can snoop the commands using the kernel's trace events. For admin commands like this one, you can set it up like:

```
# echo "qid == 0" > /sys/kernel/debug/tracing/events/nvme/filter
# echo 1 > /sys/kernel/debug/tracing/events/nvme/enable
```

The broken command:

```
# cat /sys/kernel/debug/tracing/trace
...
           <...>-3165246 [050] .... 5609217.070114: nvme_setup_cmd: nvme0: qid=0, cmdid=10, nsid=0, flags=0x0, meta=0x0, cmd=(0x1c cdw10=08 01 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00)
```

And verifying the fix:

```
# cat /sys/kernel/debug/tracing/trace
...
           <...>-3165405 [055] .... 5609536.344377: nvme_setup_cmd: nvme0: qid=0, cmdid=7, nsid=0, flags=0x0, meta=0x0, cmd=(0x1c cdw10=08 01 01 00 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00)

```

closed 2021-05-21 by keithbusch

ganaidu707, 2021-05-21:
@keithbusch I think this kind of fix required in other places like example `get_endurance_log -> group id, get_log -> LID,  get_feature, set_feature -> FID ?`

daiaji, 2021-08-09:
```
nvme virt-mgmt /dev/nvme0 -c 1 -r 0 -n 2 -a 8
success, Number of Controller Resources Modified (NRM):0
nvme virt-mgmt /dev/nvme0 -c 1 -r 1 -n 2 -a 8
success, Number of Controller Resources Modified (NRM):0
nvme virt-mgmt /dev/nvme0 -c 1 -a 9
NVMe status: INVALID_SECONDARY_CTRL_STATE: The action requested for the secondary controller is invalid based on the current state of the secondary controller and its primary controller.(0x120)
nvme version
nvme version 1.14
```
Is this a return?