linux-nvme/nvme-cli #1052: "submit-io: Invalid argument" means?

minimm, 2021-05-24:
I try to issue multiple write commands to Sony CF drive on ubuntu 18.045 LTS, The results are as follows: sometimes it works, sometimes it doesn't .  What does "submit-io: Invalid argument" mean ?  Thanks.
MDTS=2M
nvme version: nvme-1.9

```
root@ssd-X570GT:~# nvme write /dev/nvme0n1 -s 1310720 -c 4095 -d wdata -z 2097152
submit-io: Invalid argument
root@ssd-X570GT:~# nvme write /dev/nvme0n1 -s 1310720 -c 4095 -d wdata -z 2097152
write: Success
root@ssd-X570GT:~# nvme write /dev/nvme0n1 -s 1310720 -c 4095 -d wdata -z 2097152
submit-io: Invalid argument
root@ssd-X570GT:~# nvme write /dev/nvme0n1 -s 1310720 -c 4095 -d wdata -z 2097152
submit-io: Invalid argument
root@ssd-X570GT:~# nvme write /dev/nvme0n1 -s 1310720 -c 4095 -d wdata -z 2097152
write: Success
```

keithbusch, 2021-05-24:
The error is coming from the kernel. Most likely reason is that the allocated memory is sufficiently fragmented such that the kernel block layer can't map it.

The only way userspace can guarantee physically contiguous memory is with hugetlb. If you set that up and have libhugetlbfs, this program will allocate that memory for large transfers.

minimm, 2021-05-25:
Hi keithbusch,

Thank you very much for your reply, I'll try the method you said.

Thanks.

rw1nkler, 2021-11-25:
Hi @keithbusch.

Recently, I encountered the same problem and tried using the hugetlbfs but without any success. Could you tell me if my approach is correct?

### My approach

To configure the hugetlbfs, I followed the instructions from the [Debian Wiki](https://wiki.debian.org/Hugepages):

- Created a hugetlbfs group and added both the current user and the root to the group.
- Added 15 pages for the created group in the `/etc/sysctl.conf` file
- Created mount point and proper entry in `/etc/fstab` file
- Increased hard and soft `memlock` limits for both the root and the current user
- Added the following kernel boot arguments: `default_hugepagesz=1GB hugepagesz=1GB hugepages=15`

After that I saw the following information about hugetblfs:
```
╭─rwinkler@rwinkler-lenovo:~
╰─$ hugeadm --pool-list
      Size  Minimum  Current  Maximum  Default
1073741824       15       15       15        *
╭─rwinkler@rwinkler-lenovo:~
╰─$ grep Huge /proc/meminfo 
AnonHugePages:         0 kB
ShmemHugePages:        0 kB
FileHugePages:         0 kB
HugePages_Total:      15
HugePages_Free:       15
HugePages_Rsvd:        0
HugePages_Surp:        0
Hugepagesize:    1048576 kB
Hugetlb:        15728640 kB
╭─rwinkler@rwinkler-lenovo:~
╰─$ grep -R "" /sys/kernel/mm/hugepages/ /proc/sys/vm/*huge*
/sys/kernel/mm/hugepages/hugepages-1048576kB/free_hugepages:15
/sys/kernel/mm/hugepages/hugepages-1048576kB/resv_hugepages:0
/sys/kernel/mm/hugepages/hugepages-1048576kB/surplus_hugepages:0
/sys/kernel/mm/hugepages/hugepages-1048576kB/nr_hugepages_mempolicy:15
/sys/kernel/mm/hugepages/hugepages-1048576kB/nr_hugepages:15
/sys/kernel/mm/hugepages/hugepages-1048576kB/nr_overcommit_hugepages:0
/proc/sys/vm/hugetlb_shm_group:1001
/proc/sys/vm/nr_hugepages:15
/proc/sys/vm/nr_hugepages_mempolicy:15
/proc/sys/vm/nr_overcommit_hugepages:0
```
Later I installed the libhugetlbfs, which was visible for a linker. Then I compiled (and installed) the `nvme-cli`:
```
╭─rwinkler@rwinkler-lenovo:~/Programs/nvme-cli <master*>
╰─$ ld  -o /dev/null -lhugetlbfs
ld: warning: cannot find entry symbol _start; not setting start address
╭─rwinkler@rwinkler-lenovo:~/Programs/nvme-cli <master*>
╰─$ ld  -o /dev/null -lwronglibrary
ld: cannot find -lwronglibrary
```
When I'm using the `nvme-cli`, the number of free pages decreases from 15 to 14.
Even with that changes, the read/write commands are failing from time to time:
```
╭─rwinkler@rwinkler-lenovo~
╰─$ while true; do sudo nvme read /dev/nvme0n1 --start-block=0 --block-count=1024 --data-size=524800 > /dev/null; sleep 0.1; done
read: Success
read: Success
submit-io: Invalid argument
read: Success
read: Success
read: Success
submit-io: Invalid argument
read: Success
submit-io: Invalid argument
submit-io: Invalid argument
```
```
[... 40 lines cut ...]
            nvme-15668 [003]  1345.522046: function:                                              blk_queue_enter           
            nvme-15668 [003]  1345.522046: function:                                              blk_mq_get_request        
            nvme-15668 [003]  1345.522046: function:                                                 blk_mq_get_tag
            nvme-15668 [003]  1345.522047: function:                                                    __blk_mq_get_tag
            nvme-15668 [003]  1345.522047: function:                                                 ktime_get
            nvme-15668 [003]  1345.522047: function:                                                 dd_prepare_request
            nvme-15668 [003]  1345.522048: function:                                              blk_queue_exit                 
            nvme-15668 [003]  1345.522048: function:                                        blk_rq_map_user                                                    
            nvme-15668 [003]  1345.522048: function:                                           blk_rq_map_user_iov                                             
            nvme-15668 [003]  1345.522048: function:                                              bio_copy_user_iov
           ...
            nvme-15668 [003]  1345.522313: function:                                        blk_mq_free_request                                                
            nvme-15668 [003]  1345.522313: function:                                           dd_finish_request                                               
            nvme-15668 [003]  1345.522313: function:                                           __rq_qos_done                                                   
            nvme-15668 [003]  1345.522314: function:                                              wbt_done                                                     
            nvme-15668 [003]  1345.522314: function:                                           __blk_mq_free_request                                           
            nvme-15668 [003]  1345.522314: function:                                              blk_mq_put_tag                                               
            nvme-15668 [003]  1345.522314: function:                                              blk_mq_sched_restart                                         
            nvme-15668 [003]  1345.522314: function:                                              blk_queue_exit                                               
            nvme-15668 [003]  1345.522314: function:                                     nvme_passthru_end  
```
The trace above shows that, the driver returns `Invalid argument` after failing on the following line of the kernel code (SHA points to my version of the kernel):
https://github.com/torvalds/linux/blob/c6bf0ed9d1a727fcd98f6a52bbdd2afbdfb44924/drivers/nvme/host/core.c#L503

Do you have any ideas, or advice on how to debug the problem further?

keithbusch, 2021-11-25:
That error is usually associated with too many page fragments. Are you able to verify if allocating a huge page was successful? In the failed case? I can't tell from the trace output.

rw1nkler, 2021-11-26:
The trace shows, that directly after the ` blk_rq_map_user` function, the control jumps to `blk_mq_free_request`. This indicates that the [blk_rq_map_user](https://github.com/torvalds/linux/blob/c6bf0ed9d1a727fcd98f6a52bbdd2afbdfb44924/drivers/nvme/host/core.c#L937) function failed and the [goto instruction](https://github.com/torvalds/linux/blob/c6bf0ed9d1a727fcd98f6a52bbdd2afbdfb44924/drivers/nvme/host/core.c#L970) was used.

rw1nkler, 2021-11-26:
To provide more information from the application site I added [printf statements](https://github.com/rw1nkler/nvme-cli/commit/5acd8ac5e0e629da7dc3fb4955c0932f39d714a1) to the `nvme-cli` source code. Here are the results of running the application. In both cases, the app was able to allocate huge pages:
```
╭─rwinkler@rwinkler-lenovo:~/Programs/nvme-cli <master>       
╰─$ sudo ./nvme write /dev/nvme1n1 --start-block=0 --block-count=1024 --data-size=524800 --data=/dev/urandom;
submit_io:5645: before allocation       
nvme_alloc:150: get_hugepage_region succeeded            
submit_io:5652: after allocation                                                                                                                               
submit_io:5732: nvme_write, err = -1, errno = 22                               
submit-io: Invalid argument
╭─rwinkler@rwinkler-lenovo:~/Programs/nvme-cli <master>
╰─$ sudo ./nvme write /dev/nvme1n1 --start-block=0 --block-count=1024 --data-size=524800 --data=/dev/urandom;
submit_io:5645: before allocation
nvme_alloc:150: get_hugepage_region succeeded
submit_io:5652: after allocation
submit_io:5732: nvme_write, err = 0, errno = 0
write: Success
```

keithbusch, 2021-11-26:
> The trace shows, that directly after the ` blk_rq_map_user` function, the control jumps to `blk_mq_free_request`.

Sure, but that function calls many other functions that can fail. Usually it's because of page fragmentation exceeding block layer max segments. But if you're definitely getting huge pages and it's still failing, then it should be for a different reason.

igaw, 2022-01-19:
I think the question got answered. Closing the issue.

closed 2022-01-19 by igaw