linux-nvme/nvme-cli #1066: nvme flush /dev/nvme0n1 failed

yizhanglinux, 2021-06-01:
Hi Keith
I reproduced this issue again with the latest v5.13-rc4, so filed this issue in case we missed it.

BTW, I've filed it to linux-nvme@ before:

https://lore.kernel.org/linux-nvme/20210518144249.GE2709569@dhcp-10-100-145-180.wdc.com/T/#t

[FAIL]nvme flush /dev/nvme0n1
STDOUT:
STDERR:flush: Invalid argument
RETURN:22

Thanks
Yi

keithbusch, 2021-06-01:
Thank you for the reminder, I pushed an appropriate fix to account for the new kernel behavior.

yizhanglinux, 2021-06-01:
> Thank you for the reminder, I pushed an appropriate fix to account for the new kernel behavior.
With the fix, flush nvme char device will be failed, is that expected?

# ./nvme flush /dev/nvme0n1
NVMe Flush: success

# ./nvme flush /dev/nvme0
get-namespace-id: Inappropriate ioctl for device

ganaidu707, 2021-06-01:
I think if it is char device we have to specify the `nsid` 
`nvme flush /dev/nvme0 -n <nsid>`

keithbusch, 2021-06-01:
Correct, you'd have to specify -n 0xffffffff to do all namespaces through the admin handle char dev.

But we can potentially forget about maintaining this command through the character device. The kernel deprecated IO through it a while ago, and it will not even work at all if you have 2 or more attached namespaces.

yizhanglinux, 2021-06-02:
> Correct, you'd have to specify -n 0xffffffff to do all namespaces through the admin handle char dev.
>
> But we can potentially forget about maintaining this command through the character device. The [...]

Yeah, I'm thinking if we can add some warn msg when we specifying nvme char device without nsid.
Anyway since it was deprecated IO now, I'm fine with it, thanks for the explanation.

Close this issue now.

closed 2021-06-02 by yizhanglinux