linux-nvme/nvme-cli #1091: nvme list -d requires trailing /

gcoakes, 2021-06-22:
The `nvme list` command for character devices at different locations (such as spdk) fails when there is no trailing slash. i.e.:
``` bash
nvme list -d /dev/spdk  # Fails
nvme list -d /dev/spdk/ # Passes
```

This isn't a hugely important issue. Just figured I would report it so that other people don't bang their heads against the desk like I did. If this is intended behavior, it should be called out in the help text.

hreinecke, 2022-01-16:
Please retest with the latest version. I haven't been able to reproduce this issue.

igaw, 2022-02-10:
nvme-cli 2.0 doesn't have the --directory option anymore. Looking at the current code I am not sure if we actually need it, as we scan the /sys/class/nvme hierarchy. I suppose SPDK device should show up there, no?

igaw, 2022-02-10:
Obviously, the nvme-cli 1.x branch might still be affected.

igaw, 2022-02-15:
Closing this issue. Reopen if the problem still exist.

closed 2022-02-15 by igaw