linux-nvme/nvme-cli #1093: nvme list , nvme show-regs segmentation error on kernel  5.8.0-55-generic #62~20.04.1-Ubuntu

moshe-y, 2021-06-28:
dmesg printouts due to nvme-list when resulting in segmentation error.

segfault at 0 ip 0000556e75aa3230 sp 00007ffd37371d48 error 6 in nvme
======
[556e75a7d000+51000]

strace nvme list /dev/nvme0
execve("/usr/sbin/nvme", ["nvme", "list", "/dev/nvme0"], 0x7fff5b19d0b0 /* 30 vars */) = 0
brk(NULL) = 0x55d4e3475000
arch_prctl(0x3001 /* ARCH_??? */, 0x7fffa158f710) = -1 EINVAL (Invalid argument)
access("/etc/ld.so.preload", R_OK) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=117266, ...}) = 0
mmap(NULL, 117266, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7ff3e5091000
close(3) = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libuuid.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\200%\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=30936, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7ff3e508f000
mmap(NULL, 32792, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7ff3e5086000
mmap(0x7ff3e5088000, 16384, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2000) = 0x7ff3e5088000
mmap(0x7ff3e508c000, 4096, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x6000) = 0x7ff3e508c000
mmap(0x7ff3e508d000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x6000) = 0x7ff3e508d000
close(3) = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\360q\2\0\0\0\0\0"..., 832) = 832
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
pread64(3, "\4\0\0\0\20\0\0\0\5\0\0\0GNU\0\2\0\0\300\4\0\0\0\3\0\0\0\0\0\0\0", 32, 848) = 32
pread64(3, "\4\0\0\0\24\0\0\0\3\0\0\0GNU\0\t\233\222%\274\260\320\31\331\326\10\204\276X>\263"..., 68, 880) = 68
fstat(3, {st_mode=S_IFREG|0755, st_size=2029224, ...}) = 0
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
pread64(3, "\4\0\0\0\20\0\0\0\5\0\0\0GNU\0\2\0\0\300\4\0\0\0\3\0\0\0\0\0\0\0", 32, 848) = 32
pread64(3, "\4\0\0\0\24\0\0\0\3\0\0\0GNU\0\t\233\222%\274\260\320\31\331\326\10\204\276X>\263"..., 68, 880) = 68
mmap(NULL, 2036952, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7ff3e4e94000
mprotect(0x7ff3e4eb9000, 1847296, PROT_NONE) = 0
mmap(0x7ff3e4eb9000, 1540096, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x25000) = 0x7ff3e4eb9000
mmap(0x7ff3e5031000, 303104, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x19d000) = 0x7ff3e5031000
mmap(0x7ff3e507c000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1e7000) = 0x7ff3e507c000
mmap(0x7ff3e5082000, 13528, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7ff3e5082000
close(3) = 0
mmap(NULL, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7ff3e4e91000
arch_prctl(ARCH_SET_FS, 0x7ff3e4e91780) = 0
mprotect(0x7ff3e507c000, 12288, PROT_READ) = 0
mprotect(0x7ff3e508d000, 4096, PROT_READ) = 0
mprotect(0x55d4e152c000, 8192, PROT_READ) = 0
mprotect(0x7ff3e50db000, 4096, PROT_READ) = 0
munmap(0x7ff3e5091000, 117266) = 0
openat(AT_FDCWD, "/usr/lib/locale/locale-archive", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=5699248, ...}) = 0
mmap(NULL, 5699248, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7ff3e4921000
close(3) = 0
brk(NULL) = 0x55d4e3475000
brk(0x55d4e3496000) = 0x55d4e3496000
openat(AT_FDCWD, "/sys/class/nvme-subsystem/", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 3
fstat(3, {st_mode=S_IFDIR|0755, st_size=0, ...}) = 0
getdents64(3, /* 3 entries */, 32768) = 80
getdents64(3, /* 0 entries */, 32768) = 0
close(3) = 0
openat(AT_FDCWD, "/sys/class/nvme-subsystem/nvme-subsys0/subsysnqn", O_RDONLY) = 3
read(3, "nqn.2020-09.com.example:nvme:nvm"..., 256) = 78
close(3) = 0
openat(AT_FDCWD, "/sys/class/nvme-subsystem/nvme-subsys0", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 3
fstat(3, {st_mode=S_IFDIR|0755, st_size=0, ...}) = 0
getdents64(3, /* 12 entries */, 32768) = 368
getdents64(3, /* 0 entries */, 32768) = 0
close(3) = 0
openat(AT_FDCWD, "/sys/class/nvme-subsystem/nvme-subsys0/nvme0/address", O_RDONLY) = 3
read(3, "0000:01:00.0\n", 1024) = 13
close(3) = 0
openat(AT_FDCWD, "/sys/class/nvme-subsystem/nvme-subsys0/nvme0/transport", O_RDONLY) = 3
read(3, "pcie\n", 1024) = 5
close(3) = 0
openat(AT_FDCWD, "/sys/class/nvme-subsystem/nvme-subsys0/nvme0/state", O_RDONLY) = 3
read(3, "live\n", 1024) = 5
close(3) = 0
openat(AT_FDCWD, "/sys/class/nvme-subsystem/nvme-subsys0/nvme0/hostnqn", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/sys/class/nvme-subsystem/nvme-subsys0/nvme0/hostid", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/sys/class/nvme-subsystem/nvme-subsys0/nvme0", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 3
fstat(3, {st_mode=S_IFDIR|0755, st_size=0, ...}) = 0
getdents64(3, /* 21 entries */, 32768) = 664
getdents64(3, /* 0 entries */, 32768) = 0
close(3) = 0
openat(AT_FDCWD, "/sys/class/nvme-subsystem/nvme-subsys0/nvme0/nvme0c0n1/nsid", O_RDONLY) = 3
read(3, "1\n", 16) = 2
openat(AT_FDCWD, "/dev/nvme0c0n1", O_RDONLY) = -1 ENOENT (No such file or directory)
close(3) = 0
openat(AT_FDCWD, "/dev/nvme0", O_RDONLY) = 3
ioctl(3, NVME_IOCTL_ADMIN_CMD, 0x7fffa158f1f0) = 0
close(3) = 0
openat(AT_FDCWD, "/sys/class/nvme-subsystem/nvme-subsys0", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 3
fstat(3, {st_mode=S_IFDIR|0755, st_size=0, ...}) = 0
getdents64(3, /* 12 entries */, 32768) = 368
getdents64(3, /* 0 entries */, 32768) = 0
close(3) = 0
openat(AT_FDCWD, "/dev/nvme0n1", O_RDONLY) = 3
fstat(3, {st_mode=S_IFBLK|0660, st_rdev=makedev(0x103, 0x3), ...}) = 0
ioctl(3, NVME_IOCTL_ID, 0x55d4e1530cc0) = 1
ioctl(3, NVME_IOCTL_ADMIN_CMD, 0x7fffa158f1c0) = 0
close(3) = 0
fstat(1, {st_mode=S_IFCHR|0600, st_rdev=makedev(0x88, 0), ...}) = 0
write(1, "Node SN "..., 142Node SN Model Namespace Usage Format FW Rev
) = 142
write(1, "---------------- --------------"..., 142--------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
) = 142
stat("/dev/nvme0c0n1", 0x7fffa158efd0) = -1 ENOENT (No such file or directory)
stat("/dev/nvme0n1", {st_mode=S_IFBLK|0660, st_rdev=makedev(0x103, 0x3), ...}) = 0
write(1, "/dev/nvme0n1 DEADBEEF "..., 142/dev/nvme0n1 DEADBEEF PM8627 1 0.00 B / 875.89 MB 1 B + 0 B 4.0.12.x
) = 142
— SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=NULL} —
+++ killed by SIGSEGV (core dumped) +++
Segmentation fault (core dumped)

cat /sys/class/nvme/nvme0/device/resource
0x00000000a1400000 0x00000000a140ffff 0x0000000000140204
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x00000000a1000000 0x00000000a13fffff 0x0000000000140204
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000

moshe-y, 2021-06-28:
reminds a lot of 
https://github.com/linux-nvme/nvme-cli/issues/582

moshe-y, 2021-06-28:
strace nvme show-regs /dev/nvme0
execve("/usr/sbin/nvme", ["nvme", "show-regs", "/dev/nvme0"], 0x7ffea165f160 /* 30 vars */) = 0
brk(NULL) = 0x5645d467e000
arch_prctl(0x3001 /* ARCH_??? */, 0x7ffc1e733320) = -1 EINVAL (Invalid argument)
access("/etc/ld.so.preload", R_OK) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=117266, ...}) = 0
mmap(NULL, 117266, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7ff4a26ad000
close(3) = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libuuid.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\200%\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=30936, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7ff4a26ab000
mmap(NULL, 32792, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7ff4a26a2000
mmap(0x7ff4a26a4000, 16384, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2000) = 0x7ff4a26a4000
mmap(0x7ff4a26a8000, 4096, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x6000) = 0x7ff4a26a8000
mmap(0x7ff4a26a9000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x6000) = 0x7ff4a26a9000
close(3) = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\360q\2\0\0\0\0\0"..., 832) = 832
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
pread64(3, "\4\0\0\0\20\0\0\0\5\0\0\0GNU\0\2\0\0\300\4\0\0\0\3\0\0\0\0\0\0\0", 32, 848) = 32
pread64(3, "\4\0\0\0\24\0\0\0\3\0\0\0GNU\0\t\233\222%\274\260\320\31\331\326\10\204\276X>\263"..., 68, 880) = 68
fstat(3, {st_mode=S_IFREG|0755, st_size=2029224, ...}) = 0
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
pread64(3, "\4\0\0\0\20\0\0\0\5\0\0\0GNU\0\2\0\0\300\4\0\0\0\3\0\0\0\0\0\0\0", 32, 848) = 32
pread64(3, "\4\0\0\0\24\0\0\0\3\0\0\0GNU\0\t\233\222%\274\260\320\31\331\326\10\204\276X>\263"..., 68, 880) = 68
mmap(NULL, 2036952, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7ff4a24b0000
mprotect(0x7ff4a24d5000, 1847296, PROT_NONE) = 0
mmap(0x7ff4a24d5000, 1540096, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x25000) = 0x7ff4a24d5000
mmap(0x7ff4a264d000, 303104, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x19d000) = 0x7ff4a264d000
mmap(0x7ff4a2698000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1e7000) = 0x7ff4a2698000
mmap(0x7ff4a269e000, 13528, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7ff4a269e000
close(3) = 0
mmap(NULL, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7ff4a24ad000
arch_prctl(ARCH_SET_FS, 0x7ff4a24ad780) = 0
mprotect(0x7ff4a2698000, 12288, PROT_READ) = 0
mprotect(0x7ff4a26a9000, 4096, PROT_READ) = 0
mprotect(0x5645d438b000, 8192, PROT_READ) = 0
mprotect(0x7ff4a26f7000, 4096, PROT_READ) = 0
munmap(0x7ff4a26ad000, 117266) = 0
openat(AT_FDCWD, "/usr/lib/locale/locale-archive", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=5699248, ...}) = 0
mmap(NULL, 5699248, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7ff4a1f3d000
close(3) = 0
brk(NULL) = 0x5645d467e000
brk(0x5645d469f000) = 0x5645d469f000
openat(AT_FDCWD, "/dev/nvme0", O_RDONLY) = 3
fstat(3, {st_mode=S_IFCHR|0600, st_rdev=makedev(0xf2, 0), ...}) = 0
ioctl(3, NVME_IOCTL_ADMIN_CMD, 0x7ffc1e732ed0) = 1
— SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x1} —
+++ killed by SIGSEGV (core dumped) +++
Segmentation fault (core dumped)

igaw, 2022-01-19:
This is a distro kernel crash. Please report this to your distro vendor. There is nothing we can do here. nvme-cli is just the messenger.

closed 2022-01-19 by igaw