linux-nvme/nvme-cli #1094: Add shutdown sequence options to reset command

theknowledge, 2021-06-29:
***New feature request***

Add various options to nvme reset <dev> to facilitate test coverage of different shutdown sequences.
Current implementation deletes IO queue but provides no notification before issuing CC.EN=0

For full flexibility, add two parameters
- register sequences:
    - normal: issue CC.SHN=01b and poll for CSTS.SHST=10b before issuing CC.EN=0
    - abrupt: issue CC.SHN=10b then immediately CC.EN=0 (no wait for CSTS.SHST=10b)
    - no_notify: issue only CC.EN=0
- option fo IOQ delete / no delete, to be used alongside each shutdown sequence

keithbusch, 2021-06-29:
This tool doesn't actually manipulate controller registers or do the teardown sequence. It just asks the driver to do that since the driver owns the queues and register mapping.

Most Linux kernels wouldn't allow user space MMIO access while a driver is bound to the device, and even for ones that do allow it, changing them from user space will only confuse the driver when the queues stop working.

So I think what you're after is to have the driver provide these options. I'm not sure the kernel devs would want that, though.

theknowledge, 2021-06-29:
Thanks for the insight Keith.
I'm surprised the Linux kernel driver doesn't do either of the SHN options. Mind you this is CentOS (k4.18).
I'll try again with Ubuntu k5.x and see. 
If the newer kernel driver has hooks for normal and abrupt shutdown notification, would nvme-cli be able to pass out the options? Or as you say, would they not give a user mode tool those options?

keithbusch, 2021-06-29:
Linux kernel driver will use normal shutdown on S3/S4/S5 power states, notified hot removal, or manual driver unbinding. It will not use the a shutdown on a reset because most controllers will take excessively long time to restart after that, and it is of no use anyway when we already know we're not going to D3.

There is no such thing as an "abrupt shutdown" in the linux power management subsystem, so unless we get such an infrastructure in place, the driver has no particular way to be notified that an abrupt shutdown method is appropriate.

keithbusch, 2021-06-29:
Oh, and we'll also use the normal shutdown on s2idle if ACPI StorageD3 is set.

theknowledge, 2021-06-29:
OK Keith, thanks again for the detail.
Let's close this then

closed 2021-06-29 by theknowledge