linux-nvme/nvme-cli #1095: nvme format does not check holders

bdrung, 2021-06-30:
The `nvme format` command is dangerous: We use a RAID 1 with two NVMe. Sometimes a NVMe fails with a controller fatal status condition. Then we remove the faulty NVMe from the RAID, format the faulty NVMe, and add at back to the RAID for a full resync. Early in the morning this happened again, but this time I forgot to remove the faulty NVMe before formatting. Sadly this causes the filesystem on top of the RAID to choke on the formatted NVMe in the RAID. It required us to stop the consumers and do a filesystem check to get back to a clean state.

To prevent this mistake to happen again, I need a tool that checks that the NVMe is not in use before formatting. My preferred solution is that `nvme format` is enhanced instead of writing a wrapper script for checking it. I propose following change for `nvme format`:

* fail if there are holders for the nvme
* if called with `--force`, only a warning is printed
* `--ignore-holders` will not check if there are holders

Do you agree to this proposed change? Then I will start implementing it.

keithbusch, 2021-06-30:
What method do you have in mind to verify there are no holders?

bdrung, 2021-06-30:
Check the content of `/sys/block/$nvme/holders` for content. If there is content, there are holders. Otherwise it is safe to proceed.

keithbusch, 2021-06-30:
```
# mount | grep nvme
/dev/nvme0n1p1 on /boot/efi type vfat (rw,relatime,fmask=0077,dmask=0077,codepage=437,iocharset=ascii,shortname=winnt,errors=remount-ro)
# ls /sys/class/block/nvme0n1p1/holders
```

My mounted partition lists no holders, and would pass this check. That doesn't sound like what you'd want to happen. It looks like holders only lists stacking block drivers: device-mapper and md are the only subscribers to that interface, so that's only a partial solution to verifying if a namespace is in use.

The nvme device the command is typically sent to the character device rather than the block device or a partition of one. It should be possible to locate all the namespace block devices and their partitions associated with the character device's subsystem, but it's a bit messy.

But, the real problem is that multipathing these among multiple hosts. We can't really query the other hosts to see if the namespace is in use or not.

keithbusch, 2021-06-30:
Ignoring the multipath problem, I *think* just opening the affected namespace with O_EXCL should be able to verify no one is using it on the local machine. I haven't tested it yet.

bdrung, 2021-07-01:
Very good point. Let's test if opening the affected namespace with O_EXCL covers the two use cases:
* nvme used in raid
* nvme mounted

closed 2021-10-07 by hreinecke