linux-nvme/nvme-cli #1096: list in parallel with reset fails

gcoakes, 2021-06-30:
### Expectation

```
# ./nvme reset /dev/nvme0 & ./nvme list
[1] NNNNNN
Node                  SN                   Model                                    Namespace Usage                      Format           FW Rev
--------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1          SN****************** MN******************                     1          15.36  TB /  15.36  TB    512   B +  0 B   FW******
/dev/nvme1n1          SN****************** MN******************                     1          15.36  TB /  15.36  TB    512   B +  0 B   FW******
```

### Reality

```
# ./nvme reset /dev/nvme0 & ./nvme list
[1] 197102
Failed to scan namespaces
```

### Solution

Allow failure in scanning the sysfs for each device without causing the entire command to fail. Replace the data which is collected from sysfs with null in the json output and "unknown" in the normal output.

### Why

I work with nvme-cli from a firmware validation perspective. I perform reset testing on multiple devices in parallel. I would like to continue using `nvme list` without resorting to hand rolling similar behavior.

<details>
<summary>strace</summary>

```
# ./nvme reset /dev/nvme0 & strace ./nvme list
[2] 196591
execve("./nvme", ["./nvme", "list"], 0x7ffd52c25948 /* 34 vars */) = 0
arch_prctl(0x3001 /* ARCH_??? */, 0x7ffc6b2f4540) = -1 EINVAL (Invalid argument)
brk(NULL)                               = 0x1634000
brk(0x16351c0)                          = 0x16351c0
arch_prctl(ARCH_SET_FS, 0x1634880)      = 0
uname({sysname="Linux", nodename="ZP32V778-SUT15", ...}) = 0
readlink("/proc/self/exe", "/root/path/to/nvme-cli"..., 4096) = 62
brk(0x16561c0)                          = 0x16561c0
brk(0x1657000)                          = 0x1657000
mprotect(0x5c4000, 20480, PROT_READ)    = 0
openat(AT_FDCWD, "/usr/lib/locale/locale-archive", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=106075056, ...}) = 0
mmap(NULL, 106075056, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f7b866ff000
close(3)                                = 0
openat(AT_FDCWD, "/usr/share/locale/locale.alias", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=2502, ...}) = 0
read(3, "# Locale name alias data base.\n#"..., 4096) = 2502
read(3, "", 4096)                       = 0
close(3)                                = 0
openat(AT_FDCWD, "/usr/lib/locale/en_US.UTF-8/LC_TIME", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/locale/en_US.utf8/LC_TIME", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/locale/en_US/LC_TIME", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/locale/en.UTF-8/LC_TIME", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/locale/en.utf8/LC_TIME", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/locale/en/LC_TIME", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/sys/class/nvme-subsystem/", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/dev/", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 3
fstat(3, {st_mode=S_IFDIR|0755, st_size=3420, ...}) = 0
getdents64(3, /* 171 entries */, 32768) = 4984
getdents64(3, /* 0 entries */, 32768)   = 0
close(3)                                = 0
openat(AT_FDCWD, "/dev/", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 3
fstat(3, {st_mode=S_IFDIR|0755, st_size=3420, ...}) = 0
getdents64(3, /* 171 entries */, 32768) = 4984
getdents64(3, /* 0 entries */, 32768)   = 0
close(3)                                = 0
openat(AT_FDCWD, "/dev/nvme0", O_RDONLY) = -1 EAGAIN (Resource temporarily unavailable)
openat(AT_FDCWD, "/dev/nvme0n1", O_RDONLY) = 3
fstat(3, {st_mode=S_IFBLK|0660, st_rdev=makedev(259, 1), ...}) = 0
ioctl(3, NVME_IOCTL_ID, 0x5cf0c0)       = 1
ioctl(3, NVME_IOCTL_ADMIN_CMD, 0x7ffc6b2f3fd0) = 0
close(3)                                = 0
readlink("/sys/block/nvme0n1/device", "../../nvme0", 264) = 11
stat("/sys/block/nvme0n1/../../nvme0/device", {st_mode=S_IFDIR|0755, st_size=0, ...}) = 0
readlink("/sys/block/nvme0n1/../../nvme0/device", "../../../0000:8e:00.0", 264) = 21
openat(AT_FDCWD, "/dev/nvme0n1", O_RDONLY) = 3
ioctl(3, NVME_IOCTL_ADMIN_CMD, 0x7ffc6b2f2fa0) = 0
close(3)                                = 0
write(2, "Failed to scan namespaces\n", 26Failed to scan namespaces
) = 26
exit_group(-19)                         = ?
+++ exited with 237 +++
[1]   Done                    ./nvme reset /dev/nvme0
```

</details>

gcoakes, 2021-06-30:
I forgot to mention this was tested on nvme-cli v1.14, and the issue is not present in v1.8 (the version shipped with Ubuntu 20.04). I didn't narrow it down any further.

keithbusch, 2021-06-30:
Thanks for the notice on this. I will look into it this week.

gcoakes, 2021-08-23:
@keithbusch, just following up with this. If you can't prioritize it right now, does the solution I proposed sound reasonable? I might like to take this up, if it'll get accepted.

keithbusch, 2021-08-23:
I think this is corrected now in v1.15. It may not display incomplete information for the controller undergoing a reset, but it will complete scanning all namespaces available.

hreinecke, 2021-11-17:
Indeed, we had been fixing a similar issue for another customer.
So I'll be closing this issue; please open another one (against the most recent version) if the problem persists.

closed 2021-11-17 by hreinecke