linux-nvme/nvme-cli #1107: Does performing a format with the -ses=2 parameter generate a new key?

PtrStruct, 2021-07-12:
So I'm currently trying to understand this command

`sudo nvme format _device_ -ses=2`

As far as I understand, it will generate a new encryption key, essentially rendering the data that's on the drive useless.

My coworker tried it a couple of days ago on his Samsung 970 EVO and he said that once eh finished running that command he tried using a tool like R-Studio to check the Hex Editor, And the data that was previously there was now encrypted.

Would it be the same case if you were to check the data using a hardware method rather than a software method, maybe by using a PC3000?

https://github.com/linux-nvme/nvme-cli/blob/d5b822e8a7679a5892d8efa04189c7f9483e1106/Documentation/nvme-format.txt#L89

keithbusch, 2021-07-12:
The manner in which a cryptographic erase occurs is vendor specific.

PtrStruct, 2021-07-12:
Interesting, so this is something that Samsung would have to answer then, I would assume.

keithbusch, 2021-07-12:
Right, you'd have to check with Samsung. I do not know how much implementation specific details they'd be willing to share, though I think it's probably fair to say a device supports crypto erase (ID_CTRL.FNA bit 2) will irrevocably delete previous encryption keys and new one(s) will be generated. For some implementations that I'm aware of, a cryptographic erase will do a bit more than that.

PtrStruct, 2021-07-12:
Apparently `sudo nvme format _device_ -ses=2` works for the 970 EVO but not `nvme sanitize device --sanact=4`, it throws some weird opcode error 0x1 or 0x0, I can't remember the details but it was something to do with opcode, and from what I read, that's normal with Samsung drives, you would have to suspend the drive doing `systemctl suspend` I believe. Not sure if that will solve it or if it even works when the drive is not the main boot drive.

PtrStruct, 2021-09-26:
@keithbusch Does SED's require you to install a software? At least once in order to setup the controller or something similar? Or does the SED technology work straight out of the box?

The way I thought it worked was, you buy the M.2 which ships with SED, for instance the EVO 970, and then everything was implemented on the controller, where any data that gets stored in the drive first passes through the controller, gets encrypted and then decrypted realtime.

After doing some reading (and I could be wrong) it seems like you might have to install software at least once to trigger the SED function?

I remember running the nvme format with the ses=2 parameter and then checking the data and it looked encrypted. I just don't know if running that command does more than just deletes the key, does it encrypt data too? Or is the data encrypted just because of the fact that it removed the key? And no additional encryption was made?
And that's without having to install any software prior to running that command.

keithbusch, 2021-09-26:
Self encrypting drives (SED) for consumer devices will typically require a pre-boot password exchange brokered by BIOS or EFI prior to unlocking the drive. The platform should be providing the SED management interface for that case. Once unlocked and while powered on, the drive is generally usable from the host without additional SED consideration. 

The security protocol a drive uses to manage its state is outside the scope of this tool. The nvmecli tool only knows how to send and receive security payloads, but doesn't know how to interpret them.

If the platform BIOS doesn't know about your SED device, you will probably need 3rd party software to manage it.

Side note, the Linux kernel knows parts of the OPAL sed specification and can replay security sequences on resume from S3 power states.

A cryptographic format is unrelated to SED controllers. A cryptographic erase will generate a new media key, but a password exchange isn't necessary to unlock it.

If you're taking server/data center environment, then that's also gonna need 3rd party software.

igaw, 2022-01-19:
I think the question got answered. Closing the issue.

closed 2022-01-19 by igaw