linux-nvme/nvme-cli #1113: Incorrect block device assignment on attach-ns to Dual Port NVMe device

theknowledge, 2021-07-21:
*Disclaimer: not sure if this issue should be raised here, or under libnvme, or with the linux kernel itself. Your input is appreciated*

Here we have a Dual-port NVMe device:
```
root@nvme-cli# ./nvme list-subsys
nvme-subsys1 - NQN=nqn.2019-10.com.vendor:MYDEVICE:MYSERIALNUMBER
\
 +- nvme0 pcie 0000:01:00.0 live
 +- nvme1 pcie 0000:02:00.0 live
```

Controller IDs are as follows:
```
root@nvme-cli# ./nvme id-ctrl /dev/nvme0 | grep cntlid
cntlid    : 0x1
root@nvme-cli# ./nvme id-ctrl /dev/nvme1 | grep cntlid
cntlid    : 0x2
```

There are 2 namespaces in the subsystem, available to, but not attached to either controller:
```
root@nvme-cli# ./nvme list-ns /dev/nvme0
root@nvme-cli# ./nvme list-ns /dev/nvme1
root@nvme-cli# ./nvme list-ns /dev/nvme0 -a
[   0]:0x1
[   1]:0x2
root@nvme-cli# ./nvme list-ns /dev/nvme1 -a
[   0]:0x1
[   1]:0x2
```

Using the appropriate char device, attach NS1 to CTRL 0x1 and NS2 to CTRL 0x2:
```
root@nvme-cli# ./nvme attach-ns /dev/nvme0 -n1 -c1
attach-ns: Success, nsid:1
root@nvme-cli# ./nvme attach-ns /dev/nvme1 -n2 -c2
attach-ns: Success, nsid:2
```

Both are now listed under the correct char device and controller:
```
root@nvme-cli# ./nvme list-ns /dev/nvme0
[   0]:0x1
root@nvme-cli# ./nvme list-ns /dev/nvme1
[   0]:0x2
```

```
root@nvme-cli# ./nvme list-ctrl /dev/nvme0 -n1
num of ctrls present: 1
[   0]:0x1
root@nvme-cli# ./nvme list-ctrl /dev/nvme1 -n2
num of ctrls present: 1
[   0]:0x2
```

But nvme list shows both block devices are under /dev/nvme1:
```
root@nvme-cli# ./nvme list
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme1n1     MYSERIALNUMBER     MYDEVICE                                    1          0.00   B / 800.16  GB    512   B +  0 B   0101
/dev/nvme1n2     MYSERIALNUMBER     MYDEVICE                                    2          0.00   B / 799.94  GB    512   B +  0 B   0101

```

This is reflected in the device path:
```
root@nvme-cli# ls /dev/nvme*
/dev/nvme0  /dev/nvme1  /dev/nvme1n1  /dev/nvme1n2
```

So, is the tool or stack responsible for mapping namespace block devices to controller char devices??

Thanks in advance for your input.

keithbusch, 2021-07-21:
It sounds like you are using the kernel's native multipathing, and each namespace has NMIC capabilities. When set up that way, the kernel will have a top level subsystem device that inherits the name of whichever controller completed identification first. In your case, that appears to be "nvme1", so now all namespaces in that subsystem regardless of which controller(s) they are attached to will be prefixed with that.

The reason for this is so that we only need one block handle no matter how many paths you have, and the name will be consistent as you change attachment states.

I realize the way this is displayed is counter-intuitive. If you've a very recent version of this tool, it should make more sense if you run `nvme list -v`.

igaw, 2022-01-19:
So there isn't much we can do here, right?

igaw, 2022-02-15:
Closing due to inactivity. If it's relevant please reopen.

closed 2022-02-15 by igaw