linux-nvme/nvme-cli #1120: nvme create-ns return code is 22?

LiliDeng, 2021-07-28:
Hi, could you please help understand why the return code is 22, old version will return 1, thanks for your help.

```
nvme id-ctrl -H /dev/nvme0n1 | grep -i "NS Management and Attachment"
  [3:3] : 0     NS Management and Attachment Not Supported

nvme create-ns /dev/nvme0n1 --block-size 4096
NVMe status: INVALID_OPCODE: The associated command opcode field is not valid(0x1)

echo $?
22
```

keithbusch, 2021-07-28:
The exit status is an errno. It should have always been an errno. The earlier version tried to return NVMe status, but that was broken and misguided.

igaw, 2022-01-19:
The question got answered. 

As mention in #697 I think we should only use status value of 1 on error. Anything else is pretty much broken.

closed 2022-01-19 by igaw