linux-nvme/nvme-cli #1127: master branch missing commits (it look like forcely pushed)

sc108-lee, 2021-08-10:
I guess libnvme development branch pushed on master
It seems wrong, diverged with tag v1.15
@keithbusch please check it

keithbusch, 2021-08-10:
Yeah, I intended to push to a private staging tree, but had the origin set here. I was hoping to have it all fixed up and integrated correctly, but I ran out of time today. If I can't get a quick fix by early tomorrow, I'll just force push the v1.15 tag back onto master. Sorry for any breakage in flow here.

sc108-lee, 2021-08-10:
@keithbusch cool, thanks for checking fast

sc108-lee, 2021-08-10:
it haven't pushed yet, so I reopen this
After recover with v1.15 will be closed

michael-gebis, 2021-08-30:
Was this actually fixed?  My pull request (mostly just bash completions) was merged on Aug 6th, but it appears to have been wiped out since.  

https://github.com/linux-nvme/nvme-cli/pull/1121

I can easily recreate my pull request, but I'm worried other commits might have been as well.

keithbusch, 2021-08-30:
> Was this actually fixed? My pull request (mostly just bash completions) was merged on Aug 6th, but it appears to have been wiped out since.

The https://github.com/linux-nvme/nvme-cli/tree/nvme-cli-monolithic branch should have everything and will be maintained as long as needed. This was created before the forced "split" of the user/backend push happened. That integration point was unfortunately a bit older than when the split with the original master occurred, and I'm still working to resolve the divergences.

michael-gebis, 2021-08-30:
Ah, I see it now; not sure how I missed that.  Thanks for the explanation!