linux-nvme/nvme-cli #1128: Any reason to be reverted json-c bugfix?

sc108-lee, 2021-08-10:
@keithbusch 
I can see your commit(b8c7fb009f072a84632388e3f8a42092476238f0) which is revert my json-c commit(d127537694e5953551d58acf8717ab1483c38ed3).

As mentioned on my commit it cause overflow issue for json print.
And my bug fix work with json-c >= 0.14-20200419 ( I mentioned on commit as well )

There is two way to build nvme-cli. with blow option works fine (I've tested it before commit)
1. json-c >= 0.14-20200419 (with my commit)
2. without json-c (using util/json-c)

keithbusch, 2021-08-10:
I was unable to compile this with the json-c from my distro package manager, so there needs to be a new conditional compile here.

sc108-lee, 2021-08-10:
yes, I m aware of it.
centos7 latest(epel support) is 0.12
centos8 0.13
fedora 0.14
debian 11 0.15
something like this.

so I made new commit solved your worry.