linux-nvme/nvme-cli #1130: INVALID_FIELD: A reserved coded value or an unsupported value in a defined field(0x4002)

stutiredboy, 2021-08-11:
```
# nvme list
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1     PHLN1142002R1P6AGN   Dell Express Flash NVMe P4610 1.6TB SFF  1           1.60  TB /   1.60  TB    512   B +  0 B   VDV1DP24
```

`Dell Express Flash NVMe P4610 1.6TB SFF` is a Intel P4610, but the opcode may be modified by  Dell?

# nvme list-ns /dev/nvme0 -a
NVMe status: INVALID_FIELD: A reserved coded value or an unsupported value in a defined field(0x4002)

# nvme detach-ns /dev/nvme0 -n 1 -c 0
NVMe status: INVALID_OPCODE: The associated command opcode field is not valid(0x4001)

keithbusch, 2021-08-11:
The error reported by your controller says "invalid opcode", so the device is telling you that it doesn't support the optional namespace management detach operation

stutiredboy, 2021-08-11:
thanks, Dell's firmware may disable these feature.

closed 2021-08-11 by stutiredboy