linux-nvme/nvme-cli #1142: `NVMe status: Invalid Field in Command: A reserved coded value or an unsupported value in a defined field(0x2)`

paulmenzel, 2021-08-24:
$ ./nvme version
    nvme version 1.14.68.g24dd
    $ sudo ./nvme list
    Node                  SN                   Model                                    Namespace Usage                      Format           FW Rev
    --------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
    nvme0n1               BTHH82250YQK512D     SSDPEKKF512G8 NVMe INTEL 512GB           1         512.11  GB / 512.11  GB    512   B +  0 B   D03N
    $ sudo ./nvme nvm-id-ctrl /dev/nvme0n1
    NVMe status: Invalid Field in Command: A reserved coded value or an unsupported value in a defined field(0x2)

birkelund, 2021-08-24:
`nvm-id-ctrl` sends the Identify command with a `CNS` value of `0x6` ("I/O Command Set specific Identify Controller data structure") and `CSI` set to `0x0` ("NVM Command Set").

My gut feeling is that your controller does not support TP 4056 ("Namespace Types"). You can check this by inspecting the `CSS` field of the `CAP` register (see `nvme show-regs`).

paulmenzel, 2021-08-24:
```
$ sudo ./nvme show-regs /dev/nvme0n1
cap     : 30500103ff
version : 10200
cc      : 460001
csts    : 1
nssr    : 0
intms   : 0
intmc   : 0
aqa     : 1f001f
asq     : 403a978000
acq     : 403a977000
cmbloc  : 0
cmbsz   : 0
bpinfo  : 0
bprsel  : 0
bpmbl   : 0
cmbmsc	: 0
cmbsts	: 0
pmrcap  : 0
pmrctl  : 0
pmrsts  : 0
pmrebs  : 0
pmrswtp : 0
pmrmscl	: 0
pmrmscu	: 0
```

birkelund, 2021-08-24:
Add `-H` to the `show-regs` command and you get some more human-readable output ;)

But for posterity, the `CSS` field is at bits `44:37` in the CAP register, so if you take that value and shift it by `37` and mask it by `0xf` you get

    (0x30500103ff >> 37) & 0xf = 0x1

Only bit 0 is set - for Namespace Types to be supported, bit 6 must be set as well.

paulmenzel, 2021-08-24:
Thank you, indeed:

```
$ sudo ./nvme show-regs -H /dev/nvme0n1
cap     : 30500103ff
	Controller Memory Buffer Supported (CMBS): The Controller Memory Buffer is Not Supported
	Persistent Memory Region Supported (PMRS): The Persistent Memory Region is Not Supported
	Memory Page Size Maximum         (MPSMAX): 4096 bytes
	Memory Page Size Minimum         (MPSMIN): 4096 bytes
	Boot Partition Support              (BPS): No
	Command Sets Supported              (CSS): NVM command set is Supported
	                                           One or more I/O Command Sets are Not Supported
	                                           I/O Command Set is Supported
	NVM Subsystem Reset Supported     (NSSRS): Yes
	Doorbell Stride                   (DSTRD): 4 bytes
	Timeout                              (TO): 40000 ms
	Arbitration Mechanism Supported     (AMS): Weighted Round Robin with Urgent Priority Class is not supported
	Contiguous Queues Required          (CQR): Yes
	Maximum Queue Entries Supported    (MQES): 1024

version : 10200
	NVMe specification 1.2
[…]
```

paulmenzel, 2021-08-24:
> `nvm-id-ctrl` sends the `Identify` command with a CNS value of 0x6 (*I/O Command Set specific Identify Controller data structure*) and CSI set to 0x0 (*NVM Command Set*).

So, the error is expected? Is there better error handling, or is it a controller/firmware issue?

birkelund, 2021-08-24:
> > `nvm-id-ctrl` sends the `Identify` command with a CNS value of 0x6 (_I/O Command Set specific Identify Controller data structure_) and CSI set to 0x0 (_NVM Command Set_).
> 
> So, the error is expected? Is there better error handling, or is it a controller/firmware issue?

Yes. The error is expected. Your device does not know about the "I/O Command Set specific Identify Controller data structure", so it correctly responds with an "Invalid Field in Command" error (the invalid field being CNS set to `0x6`).

birkelund, 2021-08-24:
> Yes. The error is expected. Your device does not know about the "I/O Command Set specific Identify Controller data structure", so it correctly responds with an "Invalid Field in Command" error (the [...]

Are you just looking for the `id-ctrl` command maybe?

paulmenzel, 2021-08-24:
> So, the error is expected? Is there better error handling, or is it a controller/firmware issue?
>
> Yes. The error is expected. Your device does not know about the "I/O Command Set specific Identify [...]

I would have expected a different error message then. Something like “command not supported”. But now that I know, it’s fine for me.

> Are you just looking for the `id-ctrl` command maybe?

That is working. I was just testing the different subcommands.

birkelund, 2021-08-24:
> I would have expected a different error message then. Something like “command not supported”. But now that I know, it’s fine for me.

NVMe devices respond with Invalid Field in Command for both invalid commands/opcodes (the `OPC` field) as well for fields they do not understand within those commands :) `nvme-cli` cannot differentiate between those in most cases :)

paulmenzel, 2021-08-24:
Thank you for the explanation.

closed 2021-08-24 by paulmenzel

keithbusch, 2021-08-24:
There actually is an Invalid Opcode error that a controller may return. All controllers support the Identify command though, so that error is never appropriate here. This error is a command optional field that is unsupported, and Invalid Field in Command is the closest matching available status that we have.