linux-nvme/nvme-cli #1143: Print NVMe version of controller/firmware

paulmenzel, 2021-08-24:
Some properties depend on the NVMe version. Linux has the check below in `drivers/nvme/host/core.c`:

                    if (ctrl->vs >= NVME_VS(1, 2, 1))
                            dev_warn(ctrl->device, "missing or invalid SUBNQN field.\n");

How to I print the NVMe version with `nvme-cli`? Is it `ver` in

    $ ./nvme version
    nvme version 1.14.68.g24dd
    $ sudo ./nvme id-ctrl /dev/nvme0n1
    […]
    ver       : 0x10200
    […]

birkelund, 2021-08-24:
Yes, the version is packed as four bytes (`0x00010000 | 0x0200 | 0x00`) in your case. Aka, version v1.2.0.

The same information is also available in the `VS` register (see `nvme show-regs`).

closed 2021-08-24 by paulmenzel

keithbusch, 2021-08-24:
The spec compliance version is indeed noted in the Identify Controller "ver" field for 1.2 and later (the field didn't exist prior to 1.2), and in the "VS" register. However, most kernels do not support the "show-regs" command since they don't allow user space to mmap a device's IO memory space if a driver is bound to it.