linux-nvme/nvme-cli #1159: NVMe Subsystem-reset behavior

KeerthiPrabhu, 2021-09-08:
While trying to issue NVMe subsystem reset to one drive, the device gets renamed after issuing NVMe subsystem reset.

Below are the outputs of nvme list command before and after issuing NVMe subsystem reset:

Before:
# nvme list
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme2n1     SN-1       model-1									               1         691.47  GB / 960.20  GB    512   B +  0 B   FW1
/dev/nvme9n1     SN-2       model-2									               1         959.20  GB / 960.20  GB    512   B +  0 B   FW1
/dev/nvme11n1    SN-3       model-3					               				   1         691.48  GB / 960.20  GB    512   B +  0 B   FW1
/dev/nvme0n1     SN-4       model-4										           1         959.20  GB / 960.20  GB    512   B +  0 B   FW1
**/dev/nvme7n1**     SN-5       model-5					               				   1         691.47  GB / 960.20  GB    512   B +  0 B   FW1
/dev/nvme10n1    SN-6       model-6					               				   1         691.47  GB / 960.20  GB    512   B +  0 B   FW1
/dev/nvme4n1     SN-7       model-7									               1         695.75  GB / 960.20  GB    512   B +  0 B   FW1
/dev/nvme3n1     SN-8       model-8									               1         682.83  GB / 960.20  GB    512   B +  0 B   FW1
/dev/nvme5n1     SN-9       model-9									               1         691.47  GB / 960.20  GB    512   B +  0 B   FW1
/dev/nvme6n1     SN-10      model-10								               1         691.47  GB / 960.20  GB    512   B +  0 B   FW1
/dev/nvme1n1     SN-11      model-11								           	   1         691.47  GB / 960.20  GB    512   B +  0 B   FW1
/dev/nvme8n1     SN-12      model-12			    					           1         691.47  GB / 960.20  GB    512   B +  0 B   FW1

After issuing the command nvme subsystem-reset /dev/nvme7 :

# nvme list
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme2n1     SN-1       model-1									               1         691.47  GB / 960.20  GB    512   B +  0 B   FW1
/dev/nvme9n1     SN-2       model-2									               1         959.20  GB / 960.20  GB    512   B +  0 B   FW1
/dev/nvme11n1    SN-3       model-3					               				   1         691.48  GB / 960.20  GB    512   B +  0 B   FW1
/dev/nvme0n1     SN-4       model-4										           1         959.20  GB / 960.20  GB    512   B +  0 B   FW1
**/dev/nvme12n2**    SN-5       model-5					               			       1         691.47  GB / 960.20  GB    512   B +  0 B   FW1
/dev/nvme10n1    SN-6       model-6					               				   1         691.47  GB / 960.20  GB    512   B +  0 B   FW1
/dev/nvme4n1     SN-7       model-7									               1         695.75  GB / 960.20  GB    512   B +  0 B   FW1
/dev/nvme3n1     SN-8       model-8									               1         682.83  GB / 960.20  GB    512   B +  0 B   FW1
/dev/nvme5n1     SN-9       model-9									               1         691.47  GB / 960.20  GB    512   B +  0 B   FW1
/dev/nvme6n1     SN-10      model-10								               1         691.47  GB / 960.20  GB    512   B +  0 B   FW1
/dev/nvme1n1     SN-11      model-11								           	   1         691.47  GB / 960.20  GB    512   B +  0 B   FW1
/dev/nvme8n1     SN-12      model-12			    					           1         691.47  GB / 960.20  GB    512   B +  0 B   FW1

In the above output, the device seems to be renamed from /dev/nvme7n1 to /dev/nvme12n2.

nvme-cli version: 

# nvme version
nvme version 1.12
 
Can you please let me know if I am missing something here ?

Thanks !

keithbusch, 2021-09-08:
The subsystem reset causes a link down event which forces the kernel to renumerate the nvme controllers. The names are not guaranteed to be the same after that.

KeerthiPrabhu, 2021-09-08:
Thanks, @keithbusch !

In that case, there would be no way to rename the nvme controllers back to the original name ?

keithbusch, 2021-09-08:
The kernel owns the naming, there's nothing we can do about that from user space.  The kernel driver selects unique numbers for the controller names specifically to prevent naming collisions with other in-use handles.

If you want persistent names, you can find these in /dev/disk/ directory. The symlinks found under there should always be the same name for a specific controller.

KeerthiPrabhu, 2021-09-08:
Thanks, @keithbusch. Will close this issue.

closed 2021-09-08 by KeerthiPrabhu