linux-nvme/nvme-cli #1160: The maximum transfer size of a single I/O command in the shared namespace(nvme write)

ryderhsu0129, 2021-09-08:
I want to submit a single I/O command with maximum transfer size supported by the device, 
the maximum size supported by my device is 2MB, and my device also support dual-port.
If I create private namespaces to each port and submit 2MB write command to device, the command can be executed successfully;
but if I created a shared namespace and attached it to two ports, the command sometimes fail to execute.

I confirm that the cause of the fail is not from the device, It's an error from the kernel(SCT = 07h, SC = ffh), 
and the dmesg log didn't show any errors or message.
I check the relevant parameters (about maximum transfer size per command) such as max_segments(513),
seems to be enough to send the maximum transfer size by a single command.

Can anyone help clarify this problem?thanks very much!

ryderhsu0129, 2021-09-08:
@keithbusch 
Thanks for your reply.

My understanding is that max_segments=513 means that data size 2MB(513 * 4K > 2M) can be guaranteed even if the memory is not continuous.
On the other hand, you don’t think the shared namespace will affect the maximum transfer size per command?

keithbusch, 2021-09-08:
Is this a pci or fabrics target?

ryderhsu0129, 2021-09-08:
> Is this a pci or fabrics target?
pci target

keithbusch, 2021-09-08:
What kernel version are you using? PCI nvme supports a maximum of 127 segments since 4.18

ryderhsu0129, 2021-09-08:
> What kernel version are you using? PCI nvme supports a maximum of 127 segments since 4.18

5.10, but since our SSD supports I/O size of 2MB, we increase the number of segments to 513, this modification has been working well except for the applications mentioned above.

ryderhsu0129, 2021-09-08:
by the way, the LBA format is 4096 + 0

keithbusch, 2021-09-08:
What do you mean you increased max segments to 513? That is a read-only attribute, you can't change it. The value was calculated to fit in the memory that was reserved for the dma mapping. If you mean you changed the kernel to allow more segments, then I think you're on your own.

closed 2021-09-09 by ryderhsu0129