linux-nvme/nvme-cli #1170: sector size wrong when exporting a blkdev

celesteking, 2021-09-17:
Long story short,
```
# on target, an LV
NAME                ALIGNMENT MIN-IO OPT-IO PHY-SEC LOG-SEC ROTA SCHED RQ-SIZE  RA WSAME
fused-random--disk0         0   4096      0    4096     512    1           128 128    0B
# on initiator
nvme0n2         0    512      0     512     512    0 none      128 128    0B
```

As you see, minimal, optimal IO, physical sector size are all wrong. I don't care much about rotating, but I would really care about optimal io as that's a hint to queue scheduler I guess. 

There's no attribute to adjust in `/subsystems/X/namespaces/Y` to manually set the values.

Thanks.

keithbusch, 2021-09-17:
This sounds like a kernel issue in the target implementation.

celesteking, 2021-09-17:
What shall I do?

keithbusch, 2021-09-17:
The kernel mailing list for nvme kernel issues is linux-nvme@lists.infradead.org.

celesteking, 2021-09-20:
You/they should really make bug reporting user-friendly. Github's issues is one of the best tools I've ever seen.

closed 2021-09-20 by celesteking

keithbusch, 2021-09-20:
You can also file a bug against the kernel at https://bugzilla.kernel.org/enter_bug.cgi?product=IO%2FStorage

But most of the developers only subscribe to the mailing list.