linux-nvme/nvme-cli #1172: Can't find reconnection, device lost timeouts

celesteking, 2021-09-20:
How do I see the timeouts and associated params that kick in once target is lost?

`kernel: nvme nvme0: Failed reconnect attempt XX`

Thanks.

hreinecke, 2022-01-16:
That, unfortunately, is not that easy.
nvme-cli has the arguments 'ctrl-loss-tmo' and 'reconnect-delay'. 'ctrl-loss-tmo' is the maximum time the kernel will retry a connection, where each retry will be issued after 'reconnect-delay' seconds.
The kernel now translates this into the number a reconnect attempts, and stop reconnecting once 'ctrl-loss-tmo' expires.
I do agree that this is not somthing which can be easily figured out; there are default values in play (eg for ctrl-loss-tmo) which tend to confuse matters.

What we should be doing is to modify the kernel message to state:

kernel: nvme nvme0: Failed reconnect attempt XX of YY

which would make it clear where we stand.

igaw, 2022-10-28:
BTW, Linus clearly state that a syscall is must only return errno codes as error information. No string API or such:

https://lore.kernel.org/all/CAHk-=wg3cpPyoO8u+8Fu1uk86bgTUYanfKhmxMsZzvY_mV=Cxw@mail.gmail.com/

ikegami-t, 2024-05-05:
Trying to modify the kernel message as mentioned by the patch below and also add the 'ctrl-loss-tmo' description into nvme connet documention by the PR #2333.
  <https://lore.kernel.org/linux-nvme/20240505120324.20858-1-ikegami.t@gmail.com/T/#u>

closed 2024-05-07 by igaw