linux-nvme/nvme-cli #1225: nvme make failed on centos7 and ubuntu12 platform

jinhua2021, 2021-11-08:
Does nvme support centos7 and ubuntu12? It always failed with libnvme errors.
I had already submodule updated.

igaw, 2021-11-08:
Can you add some more details to your bu report? Are you building the master branch or nvme-cli-monolithic? Can you post your commands you enter and also the log output?

sc108-lee, 2021-11-24:
@jinhua2021  Centos build failure will be fixed soon :)
It will be able to build after merge blow PR
https://github.com/linux-nvme/nvme-cli/pull/1253
https://github.com/linux-nvme/libnvme/pull/131

jinhua2021, 2021-11-24:
@sc108-lee @igaw 
Thanks, waiting for that:)

igaw, 2022-01-07:
The two PR mentioned by @sc108-lee have been merged. Closing bug report. If it's still a problem, please reopen and some logs. Thanks.

closed 2022-01-07 by igaw