linux-nvme/nvme-cli #1234: Data from nvme-write not seen by a nvme-read.

gcoakes, 2021-11-11:
This is a regression between v1.14 and v1.15. `nvme write` on v1.15 appears to write zeroes only. I tested this by running `nvme write` with a known set of data (0xCE repeated for 1024 bytes) to a known set of LBAs. Then, I read back that data. On v1.14, the data read back is all 0xCE. On v1.15, the data read back is all 0x00.

`git bisect` indicates that the regression occurred in #1108.

<details>
  <summary>Log showing regression</summary>

``` console
FM21V209-FVT13 /opt/nvme-cli
# git checkout 633579877f2a860a67316e06661276c8be3ce0b1
Previous HEAD position was b9ecbc6... fix code style
HEAD is now at 6335798... Change IOCTL for read, write and compare commands

FM21V209-FVT13 /opt/nvme-cli
# make clean && make && ./nvme write /dev/nvme0n1 -s 6251233966 -c 1 -z 1024 -d /tmp/tmp.4cg6sDgpm2 && ./nvme read /dev/nvme0n1 -s 6251233966 -c 1 -z 1024 | hexdump
NVME_VERSION = 1.14.71.g6335
rm -f nvme nvme-print.o nvme-ioctl.o nvme-rpmb.o nvme-lightnvm.o fabrics.o nvme-models.o plugin.o nvme-status.o nvme-filters.o nvme-topology.o plugins/intel/intel-nvme.o plugins/amzn/amzn-nvme.o plugins/lnvm/lnvm-nvme.o plugins/memblaze/memblaze-nvme.o plugins/wdc/wdc-nvme.o plugins/wdc/wdc-utils.o plugins/huawei/huawei-nvme.o plugins/netapp/netapp-nvme.o plugins/toshiba/toshiba-nvme.o plugins/micron/micron-nvme.o plugins/seagate/seagate-nvme.o plugins/virtium/virtium-nvme.o plugins/shannon/shannon-nvme.o plugins/dera/dera-nvme.o plugins/scaleflux/sfx-nvme.o plugins/transcend/transcend-nvme.o plugins/zns/zns.o plugins/nvidia/nvidia-nvme.o plugins/ymtc/ymtc-nvme.o util/argconfig.o util/suffix.o util/parser.o util/cleanup.o util/log.o util/json.o *~ a.out NVME-VERSION-FILE *.tar* nvme.spec version control nvme-*.deb 70-nvmf-autoconnect.conf
make -C Documentation clean
make[1]: Entering directory `/opt/nvme-cli/Documentation'
rm -f *.xml *.xml+ *.html+
rm -f *.texi *.texi+ *.texi++ nvme.info nvmeman.info
rm -f *.pdf
rm -f howto-index.txt howto/*.html doc.dep
rm -f technical/*.html technical/api-index.txt
rm -f cmds-ancillaryinterrogators.txt cmds-ancillarymanipulators.txt cmds-mainporcelain.txt cmds-plumbinginterrogators.txt cmds-plumbingmanipulators.txt cmds-synchingrepositories.txt cmds-synchelpers.txt cmds-purehelpers.txt cmds-foreignscminterface.txt *.made
rm -f manpage-base-url.xsl
rm -f *~
make[1]: Leaving directory `/opt/nvme-cli/Documentation'
rm -f tests/*.pyc
rm -f verify-no-dep
NVME_VERSION = 1.14.71.g6335
    CC nvme-print.o
    CC nvme-ioctl.o
    CC nvme-rpmb.o
    CC nvme-lightnvm.o
    CC fabrics.o
    CC nvme-models.o
    CC plugin.o
    CC nvme-status.o
    CC nvme-filters.o
    CC nvme-topology.o
    CC plugins/intel/intel-nvme.o
    CC plugins/amzn/amzn-nvme.o
    CC plugins/lnvm/lnvm-nvme.o
    CC plugins/memblaze/memblaze-nvme.o
    CC plugins/wdc/wdc-nvme.o
    CC plugins/wdc/wdc-utils.o
    CC plugins/huawei/huawei-nvme.o
    CC plugins/netapp/netapp-nvme.o
    CC plugins/toshiba/toshiba-nvme.o
    CC plugins/micron/micron-nvme.o
    CC plugins/seagate/seagate-nvme.o
    CC plugins/virtium/virtium-nvme.o
    CC plugins/shannon/shannon-nvme.o
    CC plugins/dera/dera-nvme.o
    CC plugins/scaleflux/sfx-nvme.o
    CC plugins/transcend/transcend-nvme.o
    CC plugins/zns/zns.o
    CC plugins/nvidia/nvidia-nvme.o
    CC plugins/ymtc/ymtc-nvme.o
    CC util/argconfig.o
    CC util/suffix.o
    CC util/parser.o
    CC util/cleanup.o
    CC util/log.o
    CC util/json.o
    CC nvme
write: Success
read: Success
0000000 0000 0000 0000 0000 0000 0000 0000 0000
*
0000400

FM21V209-FVT13 /opt/nvme-cli
# git checkout v1.14
Previous HEAD position was 6335798... Change IOCTL for read, write and compare commands
HEAD is now at 627602c... Tag v1.14

FM21V209-FVT13 /opt/nvme-cli
# make clean && make && ./nvme write /dev/nvme0n1 -s 6251233966 -c 1 -z 1024 -d /tmp/tmp.4cg6sDgpm2 && ./nvme read /dev/nvme0n1 -s 6251233966 -c 1 -z 1024 | hexdump
NVME_VERSION = 1.14
rm -f nvme nvme-print.o nvme-ioctl.o nvme-rpmb.o nvme-lightnvm.o fabrics.o nvme-models.o plugin.o nvme-status.o nvme-filters.o nvme-topology.o plugins/intel/intel-nvme.o plugins/amzn/amzn-nvme.o plugins/lnvm/lnvm-nvme.o plugins/memblaze/memblaze-nvme.o plugins/wdc/wdc-nvme.o plugins/wdc/wdc-utils.o plugins/huawei/huawei-nvme.o plugins/netapp/netapp-nvme.o plugins/toshiba/toshiba-nvme.o plugins/micron/micron-nvme.o plugins/seagate/seagate-nvme.o plugins/virtium/virtium-nvme.o plugins/shannon/shannon-nvme.o plugins/dera/dera-nvme.o plugins/scaleflux/sfx-nvme.o plugins/transcend/transcend-nvme.o plugins/zns/zns.o plugins/nvidia/nvidia-nvme.o plugins/ymtc/ymtc-nvme.o util/argconfig.o util/suffix.o util/parser.o util/cleanup.o util/log.o util/json.o *~ a.out NVME-VERSION-FILE *.tar* nvme.spec version control nvme-*.deb 70-nvmf-autoconnect.conf
make -C Documentation clean
make[1]: Entering directory `/opt/nvme-cli/Documentation'
rm -f *.xml *.xml+ *.html+
rm -f *.texi *.texi+ *.texi++ nvme.info nvmeman.info
rm -f *.pdf
rm -f howto-index.txt howto/*.html doc.dep
rm -f technical/*.html technical/api-index.txt
rm -f cmds-ancillaryinterrogators.txt cmds-ancillarymanipulators.txt cmds-mainporcelain.txt cmds-plumbinginterrogators.txt cmds-plumbingmanipulators.txt cmds-synchingrepositories.txt cmds-synchelpers.txt cmds-purehelpers.txt cmds-foreignscminterface.txt *.made
rm -f manpage-base-url.xsl
rm -f *~
make[1]: Leaving directory `/opt/nvme-cli/Documentation'
rm -f tests/*.pyc
rm -f verify-no-dep
NVME_VERSION = 1.14
    CC nvme-print.o
    CC nvme-ioctl.o
    CC nvme-rpmb.o
    CC nvme-lightnvm.o
    CC fabrics.o
    CC nvme-models.o
    CC plugin.o
    CC nvme-status.o
    CC nvme-filters.o
    CC nvme-topology.o
    CC plugins/intel/intel-nvme.o
    CC plugins/amzn/amzn-nvme.o
    CC plugins/lnvm/lnvm-nvme.o
    CC plugins/memblaze/memblaze-nvme.o
    CC plugins/wdc/wdc-nvme.o
    CC plugins/wdc/wdc-utils.o
    CC plugins/huawei/huawei-nvme.o
    CC plugins/netapp/netapp-nvme.o
    CC plugins/toshiba/toshiba-nvme.o
    CC plugins/micron/micron-nvme.o
    CC plugins/seagate/seagate-nvme.o
    CC plugins/virtium/virtium-nvme.o
    CC plugins/shannon/shannon-nvme.o
    CC plugins/dera/dera-nvme.o
    CC plugins/scaleflux/sfx-nvme.o
    CC plugins/transcend/transcend-nvme.o
    CC plugins/zns/zns.o
    CC plugins/nvidia/nvidia-nvme.o
    CC plugins/ymtc/ymtc-nvme.o
    CC util/argconfig.o
    CC util/suffix.o
    CC util/parser.o
    CC util/cleanup.o
    CC util/log.o
    CC util/json.o
    CC nvme
write: Success
read: Success
0000000 cece cece cece cece cece cece cece cece
*
0000400

FM21V209-FVT13 /opt/nvme-cli
# hexdump /tmp/tmp.4cg6sDgpm2
0000000 cece cece cece cece cece cece cece cece
*
0000800

FM21V209-FVT13 /opt/nvme-cli
# git bisect log
git bisect start
# good: [627602c91e51052e349aac6b2c56f0cacae82648] Tag v1.14
git bisect good 627602c91e51052e349aac6b2c56f0cacae82648
# bad: [00e93879605449cabd525a85b1f8cd467e224aca] tag v1.15
git bisect bad 00e93879605449cabd525a85b1f8cd467e224aca
# good: [0ad49ade724e61f8a73e0bc78f4b190fd7085f7b] Print device name on connect
git bisect good 0ad49ade724e61f8a73e0bc78f4b190fd7085f7b
# bad: [05b5012bea348aefa9fdf5c0f065ea7ff6d7699e] nvme: add capacity management command support
git bisect bad 05b5012bea348aefa9fdf5c0f065ea7ff6d7699e
# good: [8bf6b69dc4e1759bd98bb118842aa6989c6ccac9] nvme: style fix
git bisect good 8bf6b69dc4e1759bd98bb118842aa6989c6ccac9
# good: [dacb20b4c5a88aa91dfa17829bc324565915a326] Use PRIx64 format specifier instead of lX for more portability.
git bisect good dacb20b4c5a88aa91dfa17829bc324565915a326
# bad: [633579877f2a860a67316e06661276c8be3ce0b1] Change IOCTL for read, write and compare commands
git bisect bad 633579877f2a860a67316e06661276c8be3ce0b1
# good: [b9ecbc6a1c651864840bfd16a7fe5c2135f0d9ab] fix code style
git bisect good b9ecbc6a1c651864840bfd16a7fe5c2135f0d9ab
# first bad commit: [633579877f2a860a67316e06661276c8be3ce0b1] Change IOCTL for read, write and compare commands
```
</details>

keithbusch, 2021-11-11:
Oh goodness, that is a really bad regression. I'll take a look.

keithbusch, 2021-11-11:
Ah, I think this was already fixed with bbf3beaf4d031a88aca526408d8ed94d7832a90b. I'll tag a new release since this is a pretty terrible bug to have made it into a tagged version.

keithbusch, 2021-11-11:
I've tagged a new release on the legacy branch. Please let me know if you observe any issues with that one, otherwise, I think we can close this issue.

closed 2021-11-15 by igaw