linux-nvme/nvme-cli #1239: Get-Property for offset 0x00 is only grabbing 32bits

nginsber, 2021-11-16:
Using version 1.12, when performing a  'nvme get-property  -o 0x0 /dev/nvme0n1 -H'
cap : f0003ff
        Controller Memory Buffer Supported (CMBS): The Controller Memory Buffer is Not Supported
        Persistent Memory Region Supported (PMRS): The Persistent Memory Region is Not Supported
        Memory Page Size Maximum           (MPSMAX): 4096 bytes
        Memory Page Size Minimum         (MPSMIN): 4096 bytes
        Boot Partition Support              (BPS): No
        Command Sets Supported              (CSS): NVM command set is not supported
        NVM Subsystem Reset Supported     (NSSRS): No
        Doorbell Stride                   (DSTRD): 4 bytes
        Timeout                              (TO): 7500 ms
        Arbitration Mechanism Supported     (AMS): Weighted Round Robin with Urgent Priority Class is not supported
        Contiguous Queues Required            (CQR): No
        Maximum Queue Entries Supported    (MQES): 1024

It appears only 32bits of the data is grabbed.  Offset 0x00 is 64 bits.  I know my device should be returning  0x000008200F0003FF

igaw, 2022-01-19:
Can you retest with the current master? Lot's of changes since 1.12. Thanks

igaw, 2022-02-15:
Closing the issue. Please reopen if the problem still exists.

closed 2022-02-15 by igaw