linux-nvme/nvme-cli #1241: Help converting NVME command from Windows to Linux [XPG Spectrix S40G RGB Control]

CalcProgrammer1, 2021-11-16:
This isn't specifically related to nvme-cli, but I have been using nvme-cli in my reverse engineering attempts.  This is more of me trying to understand how NVME admin commands work.

I'm the creator of OpenRGB, an open source app for RGB lighting control on PC hardware.  Our project revolves around reverse engineering the protocols that proprietary apps on Windows use to talk to RGB-lit devices, then making an open source program to control them in Windows and Linux.  We're trying to reverse engineer the XPG Spectrix S40G NVME SSD with RGB lighting.

We have the command figured out on the Windows side.  It uses STORAGE_PROTOCOL_COMMAND with DeviceIoControl to send what seems to be an NVME admin command to the device.  There seems to be some excess data at the end that contains the values to send to the RGB chip.

I've broken it down into the following command:

https://gitlab.com/CalcProgrammer1/OpenRGB/-/blob/0c8e3fd5c0934a0245970f37f09cd03b8637297a/Controllers/XPGSpectrixS40GController/XPGSpectrixS40GController.cpp#L294

```
        PNVME_COMMAND CommandValue              = (PNVME_COMMAND)command->Command;

        unsigned short corrected_reg            = ((reg << 8) & 0xFF00) | ((reg >> 8) & 0x00FF);

        CommandValue->CDW0.OPC                  = 0xFB;
        CommandValue->CDW0.CID                  = 0x0030;
        CommandValue->NSID                      = 0x00000031;
        CommandValue->u.GENERAL.CDW12           = (corrected_reg << 16) | (dev << 1);
        CommandValue->u.GENERAL.CDW13           = 0x01100001;

        DWORD ExtraValue[18] = { 0x00000000, 0x00000000, 0x00000000, 0x00000000, 0x00000000, 0x00000000, 0x00000000, 0x00000000, 0x00000000, 0x00000000,
                                 0x00000000, 0x00000000, 0x00000000, 0x00000000, 0x00000000, 0x00000000, 0x00000000, 0x00000000 };

        ExtraValue[16]                          = val;

        /*-----------------------------------------------------------------------------*\
        | Copy the ENE Register Write extra data into the STORAGE_PROTOCOL_COMMAND      |
        | buffer                                                                        |
        \*-----------------------------------------------------------------------------*/
        memcpy(&command->Command + sizeof(NVME_COMMAND), ExtraValue, sizeof(ExtraValue));

```

Opcode is 0xFB, Command ID is 0x0030, Namespace ID is 0x31, and CDW12/CDW13 are parameters.  The RGB chip is a  fairly standard one found on a lot of devices which uses I2C, so this admin command seems to be controlling an I2C master in the NVME controller.  The reg variable is a 16-bit address in the RGB chip (in this case, 0x8021, which is the RGB mode) and dev is the I2C address of the RGB chip (0x67).  The val variable is the 8-bit value to send to the RGB chip register.  The above code works well on Windows and I have no problem controlling the device.

I'm attempting to replicate it with nvme:

```
adam@Adam-Desktop:~$ sudo nvme --admin-passthru /dev/nvme0 --opcode=0xFB --rsvd=0x0030 --namespace-id=0x00000031 --cdw12=0x218000CE --cdw13=0x01100001 --input-file=test.bin -w -s -d
[sudo] password for adam: 
opcode       : fb
flags        : 00
rsvd1        : 0030
nsid         : 00000031
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00000000
metadata_len : 00000000
addr         : 0
metadata     : 0
cdw10        : 00000000
cdw11        : 00000000
cdw12        : 208000ce
cdw13        : 01100001
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 00000000
```

For the ExtraValue data I made a file called test.bin with 18 DWORD values, all 0x00000000.  If I'm understanding this correctly, it should send 0x00 for `val` in the Windows equivalent.  If I write 0x00 to the 0x8021 register in the RGB chip, the lights should turn off.  When I run the above (without the -d flag) it says NVMe command result: 0x00000000 but there is no change in the lighting, so I must have something wrong.

keithbusch, 2021-11-16:
It's always fun to hear what else is leveraging nvme as a command set and queueing mechanism. :)

I'll need to reread this a couple more times, but I just want to let you know that the "rsvd" parameter is meaningless here. If a target really requires CID=0x0030 for this operation to work, I think you're in a bit of trouble with supporting this on Linux because user space doesn't own that field, and therefore can't request it be a specific value through the inbox driver. The field will be ignored in any untainted kernel. If the target does require a specific CID, that would be a bizarre requirement for an nvme device.

keithbusch, 2021-11-16:
Odd that you specified "--cdw12=0x218000CE", but it prints out as "208000ce". Something seems off there.

On the data transfer, you specified a file to provide the payload, but you didn't specify an output size. If you're sending 18 DWORDs, the '-d' parameter needs to have value 576 (or 18 * 4).

CalcProgrammer1, 2021-11-16:
Oops, that's my fault.  I was originally putting in 208000CE when I copied the command output, but then realized I was writing to the wrong RGB chip register.  I changed it and ran the command many more times but nothing changed.  I then edited the command when I pasted it here to reflect this but forgot that the old value was in the output.

Anyways, I'm in Windows again and just set CID=0x0000 and the code is still working, so I don't think that's my problem.

keithbusch, 2021-11-16:
Ok. BTW, I mistakenly did 18 * 32 bits, but intended 18 * 4 bytes. The '-d' parameter need only be '-d 72' if you want the driver to map the payload back to user space.

CalcProgrammer1, 2021-11-16:
```
adam@Adam-Desktop:~$ sudo nvme --admin-passthru /dev/nvme0 --opcode=0xFB --namespace-id=0x00000031 --cdw12=0x218000CE --cdw13=0x01100001 --input-file=test.bin -w -s --data-len=72
opcode       : fb
flags        : 00
rsvd1        : 0000
nsid         : 00000031
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00000048
metadata_len : 00000000
addr         : 556338cc2000
metadata     : 0
cdw10        : 00000000
cdw11        : 00000000
cdw12        : 218000ce
cdw13        : 01100001
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 00000000
NVMe command result:00000000

```

Still nothing :(

keithbusch, 2021-11-16:
i can't readily see anything meaningfully different between the Windows passthrough vs the Linux attempt. Are you sure it's an admin command? I don't know the windows code at all, but the description sounds like it could be an 'io-passthru' instead.

Short of that ... do you happen to have a protocol analyzer? :)

CalcProgrammer1, 2021-11-16:
No protocol analyzer, but I do have an oscilloscope.  That might be my last resort to see that there is an I2C signal.  However, I just remembered that mode switching requires an Apply command (write 0x01 to 0x80A0) so I'm going to try that.

keithbusch, 2021-11-16:
An oscilloscope would be useless to me from an nvme protocol perspective. I figured an analyzer was a long shot, but asked anyway :)

If your new sequence doesn't work out, it may be worth trying out the io-passthru, unless you know for certain your working windows app is utilizing admin queues.

CalcProgrammer1, 2021-11-16:
I figured it out!  With nvme-cli you don't need to send the entire buffer.  There's something I don't quite understand going on with Windows between

```
        command->DataToDeviceTransferLength     = 0x00000001;
```

and

```
        command->DataToDeviceBufferOffset       = 0x000000D0;
```

The equivalent seems to be making the test.bin file only contain one byte and then using --data-len=1.  I changed my test methodology.  The RGB chip has a "direct" mode where it updates colors immediately when the direct bank of color registers is changed.  I left it in Direct mode in Windows and rebooted.  I then wrote a zero to the first byte of the Direct register bank (red channel, LED 0) and sure enough, the LED turned off!

```
adam@Adam-Desktop:~$ sudo nvme --admin-passthru /dev/nvme0 --opcode=0xFB --namespace-id=0x00000031 --cdw12=0x008100CE --cdw13=0x01100001 --input-file=zero.bin -w -s --data-len=1
[sudo] password for adam: 
opcode       : fb
flags        : 00
rsvd1        : 0000
nsid         : 00000031
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00000001
metadata_len : 00000000
addr         : 55af352af000
metadata     : 0
cdw10        : 00000000
cdw11        : 00000000
cdw12        : 008100ce
cdw13        : 01100001
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 00000000
NVMe command result:00000000
```

I'm pretty sure I can do block writes as well by making test.bin have an arbitrary number of bytes and adjusting the registers as I did in Windows.

closed 2021-11-16 by CalcProgrammer1

keithbusch, 2021-11-16:
Heh, what an obfuscated mess that appears to be. Great job figuring it out, best of luck on any future efforts. My guess is the extra data was simply padding for alignment reasons, but who knows...