linux-nvme/nvme-cli #1244: Port missing commits from nvme-cli-monolithic

igaw, 2021-11-17:
When we merged the nvme-cli-monolithic branch into master in #1223, a few commits didn't make it. Track them here:

Changes which are missing in master:

- b8a403b ("Add NVMe MI Features: Controller Metadata (0x7E) and Namespace Metadata (0x7F).")
Needs to be splitted between nvme-cli and libnvme.
Rewrite needed
include 2ae875d ("fix whitespace damage")
- 0ad49ad ("Print device name on connect")
Rewrite needed
- bace574 ("nvme-topology: fix controller check in scan_subsystem()")
Rewrite needed
- 307142b ("nvme: add get log page 2.0 spec fields")
Offset type API is missing in libnvme's nvme_get_log
- df159a7 ("nvme: add the status codes from 2.0 spec")
libnvme has the bits but the nvme.c print needs to updated accordingly
- fa97500 ("Add Timeout Parameter for Zone Management Commands")
Conflicts plugins/zns/zns.c
Changes we might need in master (either nvme-cli or libnvme)"
- 59e7477 ("nvme-print: fix 'nvme list' segfault if controller is unavailable")
not clear if we need it (does not apply)
- fc28cdc ("nvme: extend existing id ctrl parameters as per 2.0")
nvme_show_id_ctrl_ofcs() has differs a bit, not clear why.
- daa6f93 ("nvme: support ns generic device to submit_io")
no users, propably should go into libnvme

sc108-lee, 2021-11-24:
fc28cdc ofcs different reason is from missing commit 782086cc4fd590f1df171c927ebf9a2debc0a052 
so I manually recover the commits

sc108-lee, 2021-11-24:
I've reviewed df159a7 ("nvme: add the status codes from 2.0 spec")
no need any change, it's already applied well.
print also works fine via nvme_status_to_string in libnvme/src/nvme/utils.c

igaw, 2021-12-10:
Heads up, I start porting 307142b ("nvme: add get log page 2.0 spec fields")

ikegami-t, 2021-12-19:
> * [ ]  [bace574](https://github.com/linux-nvme/nvme-cli/commit/bace574bbe55739a49e7fada5483b3d3a5ef361c) ("nvme-topology: fix controller check in scan_subsystem()")
>   Rewrite needed

I seems that currently the change is not required since ctrls are continued to be scanned by nvme_subsystem_scan_ctrls() in libnvme if failed to scan namespece by nvme_ctrl_scan_namespace() in nvme_ctrl_scan_namespaces().
  Note: The 'live' controller state is not checked but it is continued to scan ctrls on the current implementation.

igaw, 2021-12-20:
> I seems that currently the change is not required since ctrls are continued to be scanned by nvme_subsystem_scan_ctrls() in libnvme if failed to scan namespece by nvme_ctrl_scan_namespace() in [...]

The orignal patch fixed the problem that 'nvme list' was showing 'Failed to open $dev' messages when a path was down (multipath NVMe setup when state was is connecting)

```
# nvme list
Failed to open /dev/nvme4
Failed to open /dev/nvme5
Failed to open /dev/nvme6
Failed to open /dev/nvme7
Failed to open /dev/nvme12
```

I can't really say what happens with the libnvme version. I try to reproduce this. It should be fairly simple (famous last words...)

igaw, 2021-12-20:
nvme list is not printing any exception. As you @ikegami-t has stated, we don't need the commit bace574 ("nvme-topology: fix controller check in scan_subsystem()")

Though there is something not correct when listing the subsystems:

```
# nvme list-subsys                                                                                                                   
nvme-subsys8 - NQN=nqn.2014-08.org.nvmexpress:NVMf:uuid:de63429f-50a4-4e03-ade6-0be27b75be77                                                                     
\                                                                                                                                                                
 +- (null) tcp traddr=10.161.8.24,trsvcid=4420 (null)                                                                                                            
 +- nvme8 tcp traddr=10.161.8.24,trsvcid=4421 live     

```
I'll open an new issue for this.

igaw, 2021-12-20:
>  59e7477 ("nvme-print: fix 'nvme list' segfault if controller is unavailable")
> not clear if we need it (does not apply)

It doesn't look we need this. I've played with my nvme-tcp test setup and blocked traffic on both paths and then I run 'nvme list', 'nvme list -v' and 'nvme list -o json' and the tool didn't crash. Let's ignore the patch for now.

igaw, 2021-12-20:
Heads up, started working on daa6f93 ("nvme: support ns generic device to submit_io")

igaw, 2021-12-21:
Heads up, started to work on a58556c ("connect-all: add -m/--matching option") and edf0998 ("fabrics: skip connect if transport type doesn't match")

hreinecke, 2021-12-21:
Watch out. '--matching' is not a great concept, and should at least by disabled per default.
I would argue for not porting it, but rather check if the port is reachable.

igaw, 2021-12-21:
> Watch out. '--matching' is not a great concept, and should at least by disabled per default. I would argue for not porting it, but rather check if the port is reachable.

Ah, okay. I just noticed these are missing. If you think we should not add them to master fine with me. In this case we just close the PR.

igaw, 2022-01-07:
Everything done on the task list. Let's close this bug report.

closed 2022-01-07 by igaw