linux-nvme/nvme-cli #1252: device-self-test not supported but self-test-log is?

birdie-github, 2021-11-22:
I've got a weird situation here:

```
# nvme device-self-test /dev/nvme0
NVMe status: INVALID_FIELD: A reserved coded value or an unsupported value in a defined field(0x6002)

# nvme self-test-log /dev/nvme0
Device Self Test Log for NVME device:nvme0
Current operation  : 0
Current Completion : 0%
Self Test Result[0-19]:
  Operation Result             : 0xf

nvme version 1.11.1
```

How is it even possible? ;-)

The device is WD_BLACK SN750 SE 1TB

jeffreyalien, 2021-11-24:
You need to specify the start test code (stc) parm for this command.  The default of stc is 0 which is not a valid value.  Here's the valid values:

- 1h Start a short device self-test operation
- 2h Start a extended device self-test operation
- eh Start a vendor specific device self-test operation
- fh abort the device self-test peration

birdie-github, 2021-11-25:
Is there documentation of that? I'm looking at `man nvme` and there aren't even any commands which are grep'pable by `self` `device` or `test`.

birdie-github, 2021-11-30:
Would be great if you mentioned `man nvme-device-self-test` in `man nvme`. Thanks!

birdie-github, 2021-11-30:
Here's one last question:

```
nvme self-test-log /dev/nvme0
Device Self Test Log for NVME device:nvme0
Current operation  : 0
Current Completion : 0%
Self Test Result[0]:
  Operation Result             : 0
  Self Test Code               : 2
  Valid Diagnostic Information : 0
  Power on hours (POH)         : 0x27
  Vendor Specific              : 0 0
Self Test Result[1]:
  Operation Result             : 0
  Self Test Code               : 1
  Valid Diagnostic Information : 0
  Power on hours (POH)         : 0x27
  Vendor Specific              : 0 0
Self Test Result[2]:
  Operation Result             : 0xf
Self Test Result[3]:
  Operation Result             : 0xf
Self Test Result[4]:
  Operation Result             : 0xf
Self Test Result[5]:
  Operation Result             : 0xf
Self Test Result[6]:
  Operation Result             : 0xf
Self Test Result[7]:
  Operation Result             : 0xf
Self Test Result[8]:
  Operation Result             : 0xf
Self Test Result[9]:
  Operation Result             : 0xf
Self Test Result[10]:
  Operation Result             : 0xf
Self Test Result[11]:
  Operation Result             : 0xf
Self Test Result[12]:
  Operation Result             : 0xf
Self Test Result[13]:
  Operation Result             : 0xf
Self Test Result[14]:
  Operation Result             : 0xf
Self Test Result[15]:
  Operation Result             : 0xf
Self Test Result[16]:
  Operation Result             : 0xf
Self Test Result[17]:
  Operation Result             : 0xf
Self Test Result[18]:
  Operation Result             : 0xf
Self Test Result[19]:
  Operation Result             : 0xf
```

This is not really nice to read.

Could you replace Self Test Code `1` and `2` with human readable strings please?
Could you replace Operation Result `0` with `OK` or `PASSED` please?

jeffreyalien, 2021-11-30:
@birdie-github I can change the output into a more human readable format and update the man page.  I won't be able get to this immediately but should be able to make the changes sometime in the next week or two.  

The power on hours number comes from the log page data returned by the drive firmware.  The nvme-cli code just reports that value.

jeffreyalien, 2021-12-01:
@birdie-github The man nvme page was updated in v1.14 to list both the device-self-test and self-test-log commands.

birdie-github, 2021-12-01:
Great many thanks! I'll leave closing this bug report to you.

igaw, 2021-12-09:
Let's close it in this case.

closed 2021-12-09 by igaw