linux-nvme/nvme-cli #1257: make install failure

engamit, 2021-11-24:
I'm using master, trying to 'make install' (after a successful 'make')

I get this error:
`install: cannot stat 'nvme-gen-dhchap-key.1': No such file or directory
install: cannot stat 'nvme-check-dhchap-key.1': No such file or directory`

I indeed see
'nvme-check-dhchap-key.txt' and 'nvme-gen-dhchap-key.txt' 
but don't see the corresponding .html and .1 files

what I am missing ?

sc108-lee, 2021-11-25:
Problem .html .1 files missing from repo

You don't miss anything for that, currently there is no file for .html and .1
It will should be generate before contributed with .txt file

will be fixed with #1259  :)

if the problem happens later, you can make doc with "make doc"
it will create .1, .html files

igaw, 2021-12-08:
#1259 has been merged, this should address the bug report here. Closing it, if there still a problem please reopen. Thanks.

closed 2021-12-08 by igaw