linux-nvme/nvme-cli #1258: nvme bash completion

clavelc, 2021-11-24:
Hi,

Bash completion seems broken for the nvme command. Pressing **tab** after typing `nvme` prints an inline prompt: 

```
[cedric@modern15 ~]$ nvme [cedric@modern15 ~]$
```

See original post on [Arch forum](https://bbs.archlinux.org/viewtopic.php?pid=2005146#p2005146)

```
$ nvme --version
nvme version 1.16

$ uname -a
Linux modern15 5.15.4-arch1-1 #1 SMP PREEMPT Sun, 21 Nov 2021 21:34:33 +0000 x86_64 GNU/Linux
```

sc108-lee, 2021-11-26:
it looks zsh(not bash completion) problem.
#1262 might be solved by this PR?

clavelc, 2021-11-26:
Hi

Thanks for your answer. I don't know about zsh, I am using Bash.

haromu, 2021-12-04:
I can confirm this error on my system.

```
$ nvme --version
nvme version 1.16

$ bash --version
GNU bash, version 5.1.12(1)-release (x86_64-pc-linux-gnu)
```

igaw, 2021-12-10:
I just merged #1271 which should fix this problem. Closing the report, if is still breaking for you, please reopen it.

PR #1271 (2021-12-10): bash: Fix nvme completion

Fixes #1258
closed 2021-12-10 by igaw