linux-nvme/nvme-cli #1260: Build fails on RHEL 8.5

mvogt1, 2021-11-25:
I tried to build current git and get the error:

make[1]: Leaving directory '/var/tmp/martin/nvme-cli/libnvme'
    CC nvme-print.o
nvme-print.c: In function ‘json_nvme_id_ns’:
nvme-print.c:164:2: error: implicit declaration of function ‘json_object_add_value_uint64’; did you mean ‘json_object_add_value_uint’? [-Werror=implicit-function-declaration]
  164 |  json_object_add_value_uint64(root, "nsze", le64_to_cpu(ns->nsze));
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |  json_object_add_value_uint
cc1: all warnings being treated as errors
make: *** [Makefile:151: nvme-print.o] Error 1

sc108-lee, 2021-11-26:
try with #1261 fix?

mvogt1, 2021-11-30:
It works even without the patch. The reason is, I though that nvme comes with its own json "util/json.h"
implemenation and uses this. But I was wrong. nvme needs a json from the system and if this fails I get the error above. This patch makes clear that a system json version is needed.
[json.txt](https://github.com/linux-nvme/nvme-cli/files/7625740/json.txt)

BarisTuncer, 2021-12-02:
I had the same issue on ubuntu, installing json-c fixed the error I got: https://github.com/json-c/json-c#buildunix

arthurshau, 2021-12-09:
I am having this issue as well on CentOS 8.
Version: json-c-0.13.1-2.el8.x86_64

sc108-lee, 2021-12-09:
@mvogt1 you were right, it caused from util/json
checkout blow command, will fix this issue
https://github.com/linux-nvme/nvme-cli/pull/1268/commits/bcd0817efa51a75b95e015f97f949806eec98e16

igaw, 2022-01-07:
As indicated by @sc108-lee this should be fixed. Closing bug report.

closed 2022-01-07 by igaw